AlexandreRouma/SDRPlusPlus #126: DSP not throttled with audio sink set to none -- bug or feature?

shuyuan-liu, 2021-04-30:
I built 96f83ee and noticed that if I set audio sink to none, the waterfall will run much faster than normal and CPU usage is consequently also higher. This could be a useful feature, for example if I wanted to process a recorded IQ file with the meteor demodulator plugin at max speed. I just wanted to make sure whether this is intended. If not, maybe consider making it a togglable feature?

AlexandreRouma, 2021-04-30:
It's intended yeah.
I use it to process recordings quickly
Might add a throttle option if people want it.

shuyuan-liu, 2021-04-30:
Thanks for confirming @AlexandreRouma, that's great! I'm fine w/o a throttling option but I wouldn't mind one being added if anyone else wants it.

closed 2021-05-13 by AlexandreRouma