AlexandreRouma/SDRPlusPlus #128: Could NOT find GLEW (missing: GLEW_INCLUDE_DIRS GLEW_LIBRARIES)

ae6jl-mdd, 2021-05-01:
I have both VS2017 and VS2019 installed. What am I missing?
[Could NOT find GLEW - missing - GLEW_INCLUDE_DIRS GLEW_LIBRARIES.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/6410280/Could.NOT.find.GLEW.-.missing.-.GLEW_INCLUDE_DIRS.GLEW_LIBRARIES.txt)

AlexandreRouma, 2021-05-02:
Hello,

you installed your packages for the wrong architecture. As you can see in vcpkg list, they all end with `:x86-windows`.

Uninstall all of those and reinstall them with the `x64-windows` instead.

PS: If you're planning on only using SDR++ and not deving, I high recommend getting the precompiled binaries from the [latest CI run](https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/803247950).

ae6jl-mdd, 2021-05-06:
Thanks. I will install the correct packages but I will also download the latest.

closed 2021-05-06 by ae6jl-mdd