AlexandreRouma/SDRPlusPlus #129: [Features] IQ file playback seeking and whole-file FFT preview

shuyuan-liu, 2021-05-02:
It would be nice if we had an overview waterfall of the IQ file currently being played back so we can do time measurements, know when short signals come up, etc. Basically what the File Player plugin for SDR# does. See https://www.rtl-sdr.com/new-sdr-plugin-file-player/ for a screenshot.

AlexandreRouma, 2021-05-02:
I was actually planning to do something similar to that plugin, just that it would pup out a window to show the entire baseband FFT and would have a cursor you could move around to go to a part of interrest.

Those changes are only planned for the rewrite of the file source tho, so it's not for very soon.

shuyuan-liu, 2021-05-02:
Thanks @AlexandreRouma, still nice to know they are planned!

janvgils, 2023-02-18:
Is there any news on the development of the file player feature requests?
It would make live so much easier when analyzing recorded signals.

kng, 2023-11-29:
I can add some info on the float32 mode. Using a Wav-sink in gnuradio-3.10 (libsndfile) to create a iq dump.
[image]
When playing this back in SDR++ I get a odd behaviour (mentioned in u8 iq file playback) where the first loop through the file has signal images and the next doesn't. This repeats with every odd/even run.
[image]
[image]

janvgils, 2024-10-17:
> I was actually planning to do something similar to that plugin, just that it would pup out a window to show the entire baseband FFT and would have a cursor you could move around to go to a part of [...]

Good morning,

Is there any news regarding this feature request?