AlexandreRouma/SDRPlusPlus #131: Audio skipping but recording is fine

shuyuan-liu, 2021-05-04:
When using SDR++ I often notice the audio "skipping" or "pausing", i.e. dropping out from time to time. The pausing is especially frequent with AM and CW, although I've also noticed it happening with NFM, USB and LSB. However any recording made with the recorder plugin is perfectly fine. See screenshots below:

[image]

[image]

Using a lower input sample rate seems to mitigate the problem a bit but it's still present. Using an RTL-SDR v3, at 2.4 MS/s the skipping in CW makes the audio output totally unusable. Interestingly, although the pauses happen at fairly regular intervals in each run, the interval is not the same every time I open SDR++. All settings are kept the same but for example in CW sometimes the audio drops out more than once every second and sometimes once every two seconds.

I'm on Arch Linux with the following software versions:
- PulseAudio 14.2
- RtAudio 5.1.0
- SDR++ 9a8150b

Hardware:
- Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz
- 8 GB RAM
- RTL-SDR v3, tested both via USB and rtl_tcp

I would be happy to provide audio recordings and / or any other information if you think they would be helpful.

AlexandreRouma, 2021-05-04:
Hello,

Firstly, the CW demod us currently broke.
Secondly, could you send a screenshot of your audio device list? I suspect RtAudio uses alsa instead of pulse which causes a lot of issues.

Regards,
Alex.

shuyuan-liu, 2021-05-04:
Hi Alex,

If you meant the audio device list in SDR++, here it is. PulseAudio is the only option in that dropdown list so I don't think Alsa is being used.

[image]

If you were asking about the PulseAudio device list, it's below. Sink 0 is my laptop's speakers, sink 1 is an Alsa loopback device. I can confirm I have set SDR++'s audio output to the speakers using `pavucontrol`.

```
Sink #0
	State: RUNNING
	Name: alsa_output.pci-0000_00_1f.3.analog-stereo
	Description: Built-in Audio Analog Stereo
	Driver: module-alsa-card.c
	Sample Specification: s32le 2ch 48000Hz
	Channel Map: front-left,front-right
	Owner Module: 6
	Mute: no
	Volume: front-left: 11796 /  18% / -44.68 dB,   front-right: 11796 /  18% / -44.68 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: alsa_output.pci-0000_00_1f.3.analog-stereo.monitor
	Latency: 104432 usec, configured 105000 usec
	Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY 
	Properties:
		alsa.resolution_bits = "32"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "ALC294 Analog"
		alsa.id = "ALC294 Analog"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0xb4318000 irq 145"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1f.3"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card1"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "9dc8"
		device.product.name = "Cannon Point-LP High Definition Audio Controller"
		device.form_factor = "internal"
		device.string = "front:1"
		device.buffering.buffer_size = "768000"
		device.buffering.fragment_size = "384000"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-stereo"
		device.profile.description = "Analog Stereo"
		device.description = "Built-in Audio Analog Stereo"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	Ports:
		analog-output-speaker: Speakers (type: Speaker, priority: 10000, availability unknown)
		analog-output-headphones: Headphones (type: Headphones, priority: 9900, availability group: Legacy 2, not available)
	Active Port: analog-output-speaker
	Formats:
		pcm

Sink #1
	State: SUSPENDED
	Name: alsa_output.platform-snd_aloop.0.analog-stereo
	Description: Built-in Audio Analog Stereo
	Driver: module-alsa-card.c
	Sample Specification: s32le 2ch 48000Hz
	Channel Map: front-left,front-right
	Owner Module: 7
	Mute: no
	Volume: front-left: 22899 /  35% / -27.40 dB,   front-right: 22899 /  35% / -27.40 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: alsa_output.platform-snd_aloop.0.analog-stereo.monitor
	Latency: 0 usec, configured 0 usec
	Flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY 
	Properties:
		alsa.resolution_bits = "32"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "Loopback PCM"
		alsa.id = "Loopback PCM"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "Loopback"
		alsa.long_card_name = "Loopback 1"
		alsa.driver_name = "snd_aloop"
		device.bus_path = "platform-snd_aloop.0"
		sysfs.path = "/devices/platform/snd_aloop.0/sound/card0"
		device.form_factor = "internal"
		device.string = "front:0"
		device.buffering.buffer_size = "768000"
		device.buffering.fragment_size = "768000"
		device.access_mode = "mmap+timer"
		device.profile.name = "analog-stereo"
		device.profile.description = "Analog Stereo"
		device.description = "Built-in Audio Analog Stereo"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output: Analog Output (type: Analog, priority: 9900, availability unknown)
	Active Port: analog-output
	Formats:
		pcm
```

Have a nice day,
Shuyuan

AlexandreRouma, 2021-05-04:
Then I have no idea what the issue is.
RtAudio seems to be causing issues just like PortAudio, but both in different ways.

shuyuan-liu, 2021-05-04:
@AlexandreRouma That's alright, thanks for the help anyway. Quite annoying that the problem doesn't happen every time, and no error messages are printed either so I really don't know where to start. I'll try sorting out the problem anyway and I'll let you know here if I make any progress.

jacobcalvert, 2021-05-26:
I've got the same issue happening here. I built from 8a18bec55. Linux Mint 20.1.

Behavior seems to be that I have clean audio for a second or two, then stuttering audio. 

I'm seeing tons of:

 RtApiAlsa::callbackEvent: audio write error, underrun.

In the console. Happy to help run this down, let me know how I can help.

jacobcalvert, 2021-06-02:
Just a follow up with some more information. I don't seem to have the problem if I connect with rtl_tcp, even over localhost. No dropouts occur in the audio for any demod setting when using rtl_tcp as the host.

jaymot123, 2021-06-30:
Click the Discussions link at the top of this GitHub page and you'll find the solution, posted by Konradrundfunk .

shuyuan-liu, 2021-08-20:
> the solution, posted by Konradrundfunk

@jaymot123 I tried the solution posted there but for some reason SDR++ wouldn't build with the custom rtaudio and I didn't bother to track down the problem. However using the new PortAudio sink seems to eliminate the audio breaks. @jacobcalvert could you confirm?

closed 2021-08-20 by shuyuan-liu