AlexandreRouma/SDRPlusPlus #132: RTL-TCP source: Remember last connected host

shuyuan-liu, 2021-05-04:
Currently RTL-TCP defaults to `localhost` every time SDR++ starts. I think it would be nice if it remembered the last address used. This would be useful if for example my SDR is connected to a Raspberry Pi and I'm using SDR++ on my computer to access it.

AlexandreRouma, 2021-05-04:
Hello,

Looks like i forgot to make it save the config when it's edited, most of the code was already there.
I'll fix it and update this when I'm done, thanks for reporting.

Regards,
Alex

mbiette, 2021-07-08:
Hello, looks like this was fixed by commit 1eca5860. Do you confirm or more work is needed on this issue?

shuyuan-liu, 2021-07-10:
Thanks @mbiette for reminding me; yes the host and port are saved now.

closed 2021-07-10 by shuyuan-liu