AlexandreRouma/SDRPlusPlus #135: Debian Sid libvolk.so.1.4 is missing

a2nt, 2021-05-06:
➜ apt search libvolk
Sorting... Done
Full Text Search... Done
libvolk2-bin/unstable,now 2.4.1-2 amd64 [installed,automatic]
  vector optimized runtime tools

libvolk2-dev/unstable,now 2.4.1-2 amd64 [installed]
  vector optimized function headers

libvolk2-doc/unstable,unstable 2.4.1-2 all
  vector optimized library documentation

libvolk2.4/unstable,now 2.4.1-2 amd64 [installed,automatic]
  vector optimized functions

~ 
➜ sdrpp
sdrpp: error while loading shared libraries: libvolk.so.1.4: cannot open shared object file: No such file or directory

AlexandreRouma, 2021-05-06:
Release 0.2.5 is old, download the latest build from here (scroll down to artifact and click on the one for debian sid)
https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/812047935

a2nt, 2021-05-06:
```
➜ uname -a
Linux darkside 5.11.0-18.1-liquorix-amd64 #1 ZEN SMP PREEMPT liquorix 5.11-32.1~sid (2021-05-02) x86_64 GNU/Linux
```

AlexandreRouma, 2021-05-06:
SoapySDR bug.
you're likely installed `soapysdr-module-all`
Uninstall anything related to soapy, then only install `libsoapysdr-dev` and the modules you need.

a2nt, 2021-05-07:
That works, thank you

closed 2021-05-07 by a2nt