AlexandreRouma/SDRPlusPlus #136: Center tuning - don't tune more than once per click

shuyuan-liu, 2021-05-07:
In center tuning mode if you click on the spectrum to change the frequency, as long as your mouse button is pressed SDR++ will keep changing the frequency. For example my current frequency is 100 MHz and I click on 99 MHz. For the duration of my mouse click, SDR++ will keep decreasing the frequency in 1 MHz steps. The result is I end up somewhere like 96 MHz instead.

AlexandreRouma, 2021-05-07:
this is on purpose but I might have to reduce the speed at which it does it

iu1mrd, 2021-08-04:
ok understood your point - it is on purpose
at the moment for me it is impossible to tune a freq that I want to tune (we are talking about "center mode")
I need some exercise to click faster ... :-) 
about the proposal of "...adding a switch..." I suggest to do this 
-- single click : tune like sdr#
-- prolonged click (press and hold pressed): change tune as long as button is pressed
btw I feel fine with "free mode" tuning

bye

AlexandreRouma, 2021-09-13:
Switched to the "move once per click" system. It should now work as expected

closed 2021-09-13 by AlexandreRouma