AlexandreRouma/SDRPlusPlus #143: Ability to set WFM Bandwidth below 50000

gr33k, 2021-05-23:
Looks like you can't set the bandwidth below 50000 for WFM which is possible in other software (SDRSharp). It would be great if this limitation was not in place to use 8000 for example.
[image]

AlexandreRouma, 2021-05-24:
hello,
this is on purpose for optimisation. Use NFM instead.

closed 2021-05-24 by AlexandreRouma

gr33k, 2021-05-24:
Hi Alexandre - I am using this with DSDPlus and NFM does not work while WFM does work. Why not let the user have the option like in SDRSharp? It works perfect in WFM but NFM produced junk in DSDPlus and no audio.

I found a temporary work around if you don't think users can benefit from this (they could as I am a user and benefit from it lol):

In radio_config.json you can edit the WFM bandwidth value (i.e. 9000). This works until you attempt to change the bandwidth which it will then again revert to the minimum of 50000.

```
        "WFM": {
            "bandwidth": 9000.0,
            "deempMode": 0,
            "snapInterval": 100.0,
            "squelchLevel": -100.0
        },
```
Hopefully you agree :)

AlexandreRouma, 2021-05-24:
I cannot allow running WFM at lower bandwidth since it needs way higher CPU usage. This limitation is to forced users to use the more optimized demodulator for the job.
Your issues with NFM are likely either the squelch, your volume or another bug.

By the way, make sure you're running the latest commit and not 0.2.5 as some demodulator bugs were fixed.

plato8888, 2023-12-19:
I agree with the gr33k.  It's difficult to perform recordings of NOAA APT with a WFM bandwidth minimum of 50Mhz when only ~38Mhz is needed.

AlexandreRouma, 2023-12-20:
Because you are NOT supposed to use WFM for NOAA APT!!!!!!!!

I don't know who keeps spreading this goddamn wrong information but WFM is meant for broadcast FM, NOT anything else.

Use NFM with IF noise reduction enabled to the NOAA APT setting.