AlexandreRouma/SDRPlusPlus #144: compile / install under MAC OSx

aneitzel, 2021-05-24:
Sorry then i use open a issue for that - but can you help in get SDR++ running on OSx please? A lot of users want to use it on MAC since SDR# was not able to do it. On platforms you list MacOS as suitable target and i waited for months now but dont see an update. 

Thanks for help!

AlexandreRouma, 2021-05-24:
Hello,

SDR++ already runs on OSX, you can download the latest build from [here](https://github.com/AlexandreRouma/SDRPlusPlus/actions?query=branch%3Amaster+event%3Apush+is%3Asuccess++)

It however has audio issues. Since I don't own a mac, I can't fix the issue myself for now.

closed 2021-05-24 by AlexandreRouma

darksidelemm, 2021-07-20:
Worth noting that the OSX builds don't seem to include the dependencies as part of the bundle, and so unless the user has installed the packages in the particular location that the sdrpp binary is expecting, the package won't work.

It looks like the sdrpp binary is expecting libraries to be present in /usr/local/opt/, e.g. /usr/local/opt/glew/lib/libGLEW.2.2.dylib. 

There is a utility out there called dylib-bundler (https://github.com/auriamg/macdylibbundler ) which packages up the dependencies for a binary and produces an app bundle, which would probably be more portable than the binary you are shipping now. The FreeDV project uses this to produce OSX App bundles: https://github.com/drowe67/freedv-gui/blob/master/src/CMakeLists.txt#L117

Unfortunately I'm kind of in a bad place with my OSX install right now (old OSX version, old gcc) so I can't really test this...

AlexandreRouma, 2021-07-20:
Yes, I prefer not including dependencies with the package. The users install what they need according to which devices they want to use.

SDR++ expects the binaries to be present wherever homebrew installs them.

I don't have a mac at all either so all the testing is done by some friends.

darksidelemm, 2021-07-20:
Adding the dependencies means the package is bigger, but is much more portable. It's a very common technique in OSX packages (which is why they are often so big) to ensure they work everywhere. 
We've had very good success doing this with FreeDV, producing packages which work on a wide range of OSX versions.