AlexandreRouma/SDRPlusPlus #145: macos audio sink not worked

BG7JAF, 2021-05-26:
Hello all:
I build the application from source and it can run up. But there is only some noise out. I look at the log message, seems that  it's something wrong with the audio buffer. Here is the log file.
By the way, recorded file is OK.

OS: macos 11.2.3
version: 0.3.0_beta

[log.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/6543454/log.txt)

porkiedev, 2021-05-26:
Bump!

I'm having this issue on Ubuntu MATE 20.04 LTS. Though I haven't built from source, I was considering trying it.
My audio sink sounds good sometimes, then it'll randomly start buffering like crazy, probably 8-10 buffers a second, so the audio is still intelligible but really annoying to listen to since it keeps freezing.

AlexandreRouma, 2021-05-26:
This is a known issue. Until I can get my hands on a real mac I can't fix this.

> I'm having this issue on Ubuntu MATE 20.04 LTS
there is no link to the OSX issue.
This issue is likely RtAudio trying to use Alsa instead of pulseaudio. You need to build RtAudio from source and disable the Alsa backend.

closed 2021-10-20 by BG7JAF