AlexandreRouma/SDRPlusPlus #151: Bonjour Alexandre créateur de SDR ++ Pouvez-vous créer une nouvelle version avec les DLL de la clé RTL  SDR

gaspar1987, 2021-06-25:


gaspar1987, 2021-06-25:
Merci Monsieur de créé DLL pour qu'on puisse avoir accès aux clés parce que sur votre forum il n'y a aucune explication pour la clé RTL je voudrais s'il vous plaît que vous me donner des instructions pour que je puisse mettre des des ailes pour faire fonctionner la clé sur votre logiciel merci cordialement

AlexandreRouma, 2021-06-25:
Bonjour,

Toutes les builds automatique récentes de SDR++ contiennent cette DLL. Je recommende d'utiliser celles ci plutot que les versions sur la page "Release" car elles sont beaucoup plus récentes.

Vous pouvez trouver ces builds [dans l'onglet "Action" de la page github](https://github.com/AlexandreRouma/SDRPlusPlus/actions?query=event%3Apush+is%3Asuccess+branch%3Amaster+actor%3AAlexandreRouma)

Cliquez sur la version la plus récente, puis descendez vers la partie "Artifacts" de la page, vous trouverais les versions pré-compilé pour tous les système d'exploitation supporté (dans votre cas sdrpp_windows_x64)

Ensuite dans l'onglet "Source" de SDR++ il suffit de choisir "RTL-SDR", votre clé devrais apparaitre dans la liste. Si elle n'apparait pas, vérifié que les drivers sont correctement installé.

closed 2022-08-17 by aosync