AlexandreRouma/SDRPlusPlus #153: Weird UI issue

jaymot123, 2021-06-30:
If I drag the horizontal row of frequencies to the right using my mouse the freqs move from left to right and get smaller as expected, but if I use my arrow keys on the keyboard the freqs move from _right to left_ as they get smaller when pressing the right-arrow key, just the opposite of what I expect which is that it should behave the same when using the keyboard as when using the mouse. They move different directions when lowering the displayed range of frequencies depending on whether I'm using my mouse or keyboard.  It's kind of confusing.

I tried and failed to record a video to show you what I mean,, sorry.

AlexandreRouma, 2021-06-30:
Not a bug.

If you use the arrow keys while hovering over the frequency scale, that'll have the same effect than dragging the frequency scale (ie left is like dragging left which will look like the frequency of VFOs go up, and vice versa)
If you use the arrow keys while on tje FFT or waterfall, it'll instead move the selected VFO. In that case pressing left will be like dragging the VFO to the left which decreases its frequency.

closed 2021-06-30 by AlexandreRouma

jaymot123, 2021-06-30:
I see. It was just confusing. I expected things to visually move in the same direction whether I used the mouse or keyboard to move them.