AlexandreRouma/SDRPlusPlus #156: Light theme - VFO barely visible; grid lines a bit too dark

shuyuan-liu, 2021-07-01:
I'm one of those people that use light themes - for eyesight reasons - and I'm happy that SDR++ has got a light theme now. When trying out the light theme I found the VFO to be not very visible however, being drawn as a quite transparent white overlay on the already white spectrum background. Besides, imo the grid lines on the spectrum look a bit too dark and "clutters up" the spectrum. I think it would be great if we could use a dark overlay for the VFO in light mode.

AlexandreRouma, 2021-07-01:
hello, the VFO color is normal, I don't change in on purpose because people might have it set to something custom.
Simplu go to the VFO color menu and set it to black.

For the lines I'll see if I can add a theme option for it

shuyuan-liu, 2021-07-01:
Ah thanks, I totally forgot about the VFO colour settings. Yes an option to change the line colours would be nice to have.

closed 2021-07-26 by AlexandreRouma