AlexandreRouma/SDRPlusPlus #158: easy linux install?

opabert, 2021-07-03:
Have new laptop and did fresh install linux.  i already did install GNUradio, Gqrx, CubicSDR.  Is there a easy to use install script for install SDR++?

AlexandreRouma, 2021-07-03:
Which distribution?

If you want to build from source, it's just two commands, check the readme for details.

If you're on debian, ubuntu or anything based on these two, .deb files are built automatically and can be downloaded this way:
1) Go to the [list of automated builds](https://github.com/AlexandreRouma/SDRPlusPlus/actions?query=event%3Apush+is%3Asuccess+branch%3Amaster+actor%3AAlexandreRouma)
2) Click on the most recent one
3) Scroll down to "Artifacts" and download the binaries for your operating system.

You can then install the deb file by running  `sudo dpkg -i <package file you downloaded>`

Make sure to first install the dependencies, they are listen in the readme

opabert, 2021-07-03:
Sorry, linux Mint 20.1 (ubuntu).  Are the dependencies already done with GNUradio or those others SDR?  Because i get hopelessly stuck while try install leading me to no ware land.

AlexandreRouma, 2021-07-04:
> Are the dependencies already done with GNUradio or those others SDR? 
I don't know. Follow the readme instructions and install the ones listed.

> i get error : dpkg file or map is not existing
I doubt that is the actual error, please copy paste the exact error you are getting.

opabert, 2021-07-04:
sudo dpkg -i sdrpp_debian_amd64.deb   is installed  from here i get stuck in "Running for development"  or "Installing SDR++"
Not all users are linux guru  and most linux install tutorials are written from developer point of view (he know what he is doing) . It  will be nice to write a simple step by step or a one line,  like a get or git instal script, install for dummys. Or make 2 install for dummys or pro. This is meanly reason some people stop using this type of experimenting programs over the past 30 years and therefore the development of this nice receiver become not popular.  To get more users using it,  a easy to fallow 1 2 3 install script.  just my opinion.

AlexandreRouma, 2021-07-04:
There is no need for an install script when all that's needed is **two** commands...

1) sudo dpkg -i sdrpp_debian_amd64.deb
2) sudo apt install -y build-essential cmake git libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libairspy-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev librtlsdr-dev libbladerf-dev liblimesuite-dev

opabert, 2021-07-04:
Done that as i normaly do, i dont see any SDR++  what i missing?

AlexandreRouma, 2021-07-04:
I don't use linux mint so not sure.
Try running `sdrpp` in a terminal.

opabert, 2021-07-04:
sdrpp: error while loading shared libraries: libvolk.so.1.4: cannot open shared object file: No such file or directory

AlexandreRouma, 2021-07-04:
You've downloaded the wrong deb file. You need to download the one for ubuntu focal. Otherwise the library versions won't match up.

opabert, 2021-07-04:
I found the link to file on this page  https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/812047935
Possible add this link in the install instructions?

AlexandreRouma, 2021-07-04:
I cannot add the link to the instructions because it changes on every build. I don't want people to be downloading old builds and experience bugs.

Currently the latest build is https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/998575350

Download the one that says "sdrpp_ubuntu_focal_amd64" and install it as mentioned last time.

opabert, 2021-07-04:
> Currently the latest build is https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/998575350

This is working  https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/812047935
Possible install this new 998575350  over the old 812047935 ?
Possible to make a start button on desktop?

AlexandreRouma, 2021-07-04:
The newer version has the start button. If linux mint doesn't show it then it's an issue with linux mint, not SDR++.
Yes you can install the new version over the old one.

opabert, 2021-07-04:
Done, up and running.  I test with  PlutoSDR, Airspymini, R2, RTL T820 + E4000 Thank you in behave of myself be happy linux dummy :)
Wait until your voice decoders plugin implemented

opabert, 2021-07-04:
BTW  use WFM  mode ,  i hear some sort of "fading" in audio but the FM radio is 500 meters distance :)

closed 2022-07-12 by AlexandreRouma