AlexandreRouma/SDRPlusPlus #161: Reoganize modules into a seperate subfolders

mbiette, 2021-07-08:
Hello,

Just a quick proposal on the organization of the files as I'm getting familiar with the project. Currently, everything is at the root of the repository which makes it hard to navigate despite the good naming convention.
I propose to move the modules into folders such as below:
- modules/sources/
- modules/decoders/
- modules/sinks/
- modules/misc/

Let me know what you think.

AlexandreRouma, 2021-07-08:
Hello,

This is already planned but I can't doi it now as I've got several people working on modules and such a change will cause a merging nightmare.

mbiette, 2021-07-08:
I agree about the directory depth, better having less than too much, that's why I sometime have trouble with Java projects structure as they usually go very deep. Here, 2 additional folder levels doesn't sound too bad.

From experience it is better to have names going from more generic to more specific. Using the naming convention you propose <type>_modules you will still get scattered folders in the root of the repository because it has a specific term at the beginning, the same way that currently there is <name>_<type> (e.g. "hackrf_source").

I tried out the structure I proposed in this branch https://github.com/mbiette/SDRPlusPlus/tree/reorg_modules if you want to check it out. Specifically in term of code change there is very little and having one or 2 levels makes no difference (apart from using a / instead of an _ in the paths)

After the change anyway you will need to ask the module developers to rebase on the most recent version of master and repush with the correct structure if they haven't done so already. Better do this kind of change earlier rather than later especially if there are more people showing interest. :)