AlexandreRouma/SDRPlusPlus #171: nlohmann::detail::parse_error

kinsha88, 2021-07-16:
Had SDR++ working for a few hours, and then it stopped running when I tried opening the program again. Running from console I get:
[2021-07-16 21:01:24.050] [info] SDR++ v0.3.0_beta
[2021-07-16 21:01:24.051] [info] Loading config
terminate called after throwing an instance of 'nlohmann::detail::parse_error'
  what():  [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
Aborted (core dumped)

Ubuntu 20.04.2 LTS
Multiple uninstalls/reinstalls/restarts does not fix or change the error, which is weird. I did not do anything outstanding before this error appeared.

AlexandreRouma, 2021-07-16:
Sounds like a corrupted config file, reinstalling doesn't replace those.

could you send the `~/.config/sdrpp/config.json` file?

Then, try deleteing the `~/.config/sdrpp` directory.

kinsha88, 2021-07-17:
The ~/.config/sdrpp/config.json is completely empty.
Deleting the directory works! Thanks

closed 2021-07-17 by kinsha88

jordancrombie, 2021-07-21:
Just for reporting purposes.
I can report that I am also getting this error:

[2021-07-20 23:43:05.494] [info] SDR++ v0.3.0_beta
[2021-07-20 23:43:05.495] [info] Loading config
[2021-07-20 23:43:06.169] [info] Loading icons
[2021-07-20 23:43:06.193] [info] Loading band plans
[2021-07-20 23:43:06.215] [info] Loading band plans color table
[2021-07-20 23:43:06.228] [info] Loading modules
[2021-07-20 23:43:06.228] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_sdr_source.dylib
[2021-07-20 23:43:06.245] [info] Loading /usr/local/lib/sdrpp/plugins/meteor_demodulator.dylib
[2021-07-20 23:43:06.262] [info] Loading /usr/local/lib/sdrpp/plugins/airspy_source.dylib
[2021-07-20 23:43:06.283] [info] Loading /usr/local/lib/sdrpp/plugins/radio.dylib
[2021-07-20 23:43:06.293] [info] Loading /usr/local/lib/sdrpp/plugins/rigctl_server.dylib
[2021-07-20 23:43:06.313] [info] Loading /usr/local/lib/sdrpp/plugins/audio_sink.dylib
[2021-07-20 23:43:06.331] [info] Loading /usr/local/lib/sdrpp/plugins/frequency_manager.dylib
[2021-07-20 23:43:06.347] [info] Loading /usr/local/lib/sdrpp/plugins/recorder.dylib
[2021-07-20 23:43:06.363] [info] Loading /usr/local/lib/sdrpp/plugins/file_source.dylib
[2021-07-20 23:43:06.379] [info] Loading /usr/local/lib/sdrpp/plugins/spyserver_source.dylib
[2021-07-20 23:43:06.396] [info] Loading /usr/local/lib/sdrpp/plugins/new_portaudio_sink.dylib
[2021-07-20 23:43:06.415] [info] Loading /usr/local/lib/sdrpp/plugins/discord_integration.dylib
[2021-07-20 23:43:06.424] [info] Loading /usr/local/lib/sdrpp/plugins/weather_sat_decoder.dylib
[2021-07-20 23:43:06.446] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_tcp_source.dylib
[2021-07-20 23:43:06.463] [info] Loading /usr/local/lib/sdrpp/plugins/hackrf_source.dylib
libc++abi: terminating with uncaught exception of type nlohmann::detail::parse_error: [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
Abort trap: 6

Base config file was fine, but the hackrf one was toast (zero bytes).
Just deleting the offending file got the startup to complete..

Also should note that in the runup to this, I was getting some strange results when "refreshing" for the hackrf on plugin...values that were not the serial number were being shown in the window.  An unplug/re-plug sorted the issue, but following that this error occured.