AlexandreRouma/SDRPlusPlus #173: Audio Problem with Pipewire and Jack

joshuarubin, 2021-07-16:
I just wanted to point out that the audio sink is unusable with the default rtaudio sink on my system. It seems similar to https://github.com/thestk/rtaudio/issues/286.

The symptom is that all other system audio stops working and a buzz on a 0.25s interval or so is played through the selected source instead of the desired audio.

I was able to work around this by recompiling `audio_sink/src/main.cpp` so that `RtAudio audio;` is changed to `RtAudio audio = RtAudio(RtAudio::LINUX_PULSE);`. I know this is not a generic solution, but I wanted to list this here in case anyone else ran into the issue.

AlexandreRouma, 2021-07-16:
Hello,

Yes this is a known issue highligned in https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112 
I will be switching back to portaudio as RtAudio doesn't seem to be a reliable library.

In the mean time this is indeed the recommended fix.

closed 2021-07-16 by AlexandreRouma

joshuarubin, 2021-07-16:
portaudio is a decent solution, but it's support for realtime, low latency, systems is pretty minimal. If you could please at least ensure that portaudio's support for jack is working when you do this, it would be much appreciated. Thanks!

AlexandreRouma, 2021-07-16:
PortAudio is used by basically all other SDR applications besides CubicSDR and a bunch of , there's no issue with latency in my experience.

i'm pretty sure portaudio already supports jack as it's listed and considering it's used by so many well known apps like VLC, Audacity, SDR#, FLdigi, etc... I'm sure support for it is pretty good.

mbiette, 2021-07-16:
PR #175  has been opened with a rework of the sink to use PortAudio. Feedback on it is welcome.

kistlin, 2022-01-26:
In addition I started implementing a PipeWire sink on [feature/add_pipewire_support](https://github.com/kistlin/SDRPlusPlus/tree/feature/add_pipewire_support). (No pipewire-pulse or -jack required)
Currently it is playing (still hard-coded) on the default device with one channel at 48kHz. Also the PipeWire header paths in the CMakeLists.txt / main.cpp most likely need to be adapted. And PipeWire >0.3 is required.

But playing some nearby radio stations works fine :).

kistlin, 2022-01-26:
Maybe thestk/rtaudio#324 adds support soon, then this could be replaced.

HT-Dat, 2024-05-27:
Changing **RtAudio audio;** to `RtAudio audio = RtAudio(RtAudio::LINUX_PULSE);` and recompile did the trick

PopOS 22.04