AlexandreRouma/SDRPlusPlus #174: Feature Request: rigctl client

joshuarubin, 2021-07-16:
I see that you have a rigctl server which is great, and I might be able to utilize this to do what I'd like. However a rigctl client module would be even better.

What I'd like to be able to do is have the frequency, mode, etc. synchronize between my HF rig so that when I change the vfo on my rig, it's reflected in sdr++ and so that when sdr++ changes the freq, my rig will follow.

Thanks! Really enjoying the software so far!

AlexandreRouma, 2021-07-16:
Should be doable.

I'll have a look after releasing the next version, trying to cut down on the todo list atm.

41south, 2021-11-29:
Hi there, I love sdr++, just wondering if Omnirig integration is in the pipeline?

AlexandreRouma, 2021-11-29:
Omnirig support is currently not planned.

TheSkorm, 2022-02-06:
This plus a way to flip IQ would be super useful for those (like me) that have IF taps on their radios.

HoracioDos, 2022-12-03:
Hi. I love SDR++. Please add rigctl client support. I would be wonderfull to change freq automatically when it it is used as a panadapter. Thank you for all your work.!

HoracioDos, 2022-12-03:
Hello!
I've found a workaround to make it work. There is a very usefull tool that allows to sync rigctl servers or rigs.
https://github.com/daveriesz/rigsync.git
Once it is compiled, you can run:
./rigsync -m 2 -r 192.168.1.42:4532 -m 2 -r 192.168.1.41:4532 -n 1

An IC7100 is attached to the first server via USB, the second one is sdrpp rigctl server with a RTL-SDR dongle.
For example: Every time I change Freq in WSJT-X (set to the first server) it changes in the IC7100 and also changes in SDR++

Hope this can help to others.

HoracioDos, 2023-02-21:
Hello. Any news on this?

AlexandreRouma, 2023-03-30:
it's being worked on but no ETA.

gerner, 2024-02-21:
I'm taking a crack at implementing bi-directional communication for my own purposes. I'm curious about the thinking with `ifFreq` and `SourceManager::TuningMode::PANADAPTER` hereish: https://github.com/AlexandreRouma/SDRPlusPlus/blame/master/core/src/signal_path/source.h#L53

it's used in tune hereish:
https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/core/src/signal_path/source.cpp#L88

It was added at the same time that rigctl_client work was getting done. It seems like it's meant to prevent tuning the SDR source while in panadapter mode, unless the ifFreq is changed. But doesn't the tuner already protect the SDR that way? hereish:
https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/core/src/gui/tuner.cpp#L56-L108

those lines make it seem like the SDR only gets tuned if the frequency falls outside the SDR's bandwidth. So why add `ifFreq` and `SourceManager::TuningMode::PANADAPTER`? Why does the tuning the SDR need to be protected this way at all? Is it just lag on tuning?

Also, `setPanadpterIF` is probably a typo and should probably be `setPanadapterIF`. There's a comment somewhere that has the same typo.

AlexandreRouma, 2024-02-21:
I'd suggest reading up on what a panadapter does and how it does it. The entire concept of a panadapter is one of the IFs of the radio is exposed on the back to allow a cheap, single frequency receiver to display the spectrum that the radio is currently receiving. 

The entire point of the panadapter tuning mode is the SDR is locked onto the real radio's IF frequency while the rest of the software assumes the currently tuned frequency to be the one selected by the user. This includes the rigctl client module which then sends this new center frequency to the real radio.

gerner, 2024-02-21:
But this implementation of `ifFreq` isn't reflecting what the radio's IF frequency is. And I don't think that rigctl exposes that information. So it's not clear to me how we'd plumb that through. In that way, we're kind of approximating what you're suggesting. And it seems like `tuner` already does that approximation.

Are you saying you're planning on using this facility (`ifFreq` and `SourceManager::TuningMode::PANADAPTER`) along with getting the radio's IF frequency some other way? I'm trying to figure out what the plan is for this code since it's not quite complete at the moment and I'm trying to fill in some of the blanks. Or I can just ignore it :)

gerner, 2024-02-21:
I should add, this is a great project, very useful for radio operators. Thank you!

gerner, 2024-02-21:
Ok, thanks for the info. If there's interest in what I'm doing, I can post back. I look forward to your progress.