AlexandreRouma/SDRPlusPlus #176: Different baseband recording precisions

sgcderek, 2021-07-18:
There seems to be no ability to select a different file type for baseband recordings, with 32-bit wav (?) being the default. This seems a bit overkill for most cases, especially for receivers such as the HackRF that are limited to 8-bit precision by hardware and thus waste a lot of disk space when recording at high sample rates.

shuyuan-liu, 2021-07-19:
I'm on commit ebd74d1d1a7ec7a7ad9200a8a74851cee2cccbd8 and with my RTL-SDR and Airspy Mini the default seems to be S16LE (16-bit values, but each sample has an I and a Q value so each sample is 32 bits). Still, I agree that recording 16-bit files from an 8-bit source is an inefficient use of disk space.

AlexandreRouma, 2021-07-20:
Hello,

This is a planned feature.

closed 2023-03-10 by AlexandreRouma