AlexandreRouma/SDRPlusPlus #177: rigctl incorrect vfo settings

joshuarubin, 2021-07-19:
When in the "fixed" waterfall mode (not sure what to call the difference between ⇆ and ⌖, but I'm talking about ⇆ mode). If I change the vfo (and the waterfall stays fixed on the screen), the frequency reported by the rigctl server does not get updated. It looks like rigctl always receives the center frequency instead. Similarly, when in ⇆ mode, the vfo can get very confused when setting frequencies (e.g. `F 14150000`) and the indicator on the waterfall can get out of position relative to what the frequency counter says.

AlexandreRouma, 2021-07-19:
hello,

I don't quite understand what you mean, could you show some screenshot (or even better screenrecording) of examples where the VFO doesn't behave as expected when tuned from rigctl?

Also please show what setting syou chose in the rigctl server

joshuarubin, 2021-07-20:
I'll see if I can get a video, but it's really easy to reproduce.
1. Start the sdr
2. Start the rigctl server
3. Switch to ⇆ mode
4. Open a connection to the rigctl server in a terminal (e.g. I use `nc localhost 4532`)
5. Check the frequency in the terminal `f`
6. Select a new frequency in the sdr by clicking the waterfall
7. Check the frequency again in the terminal, `f`, see that it hasn't changed

AlexandreRouma, 2021-07-20:
I can't replicate the issue, when I do it the frequency changes and is the same than indicated in SDR++.

Make sure you've selected the "Controlled VFO".

PS: If you added the rigctl server module through the module manager, this is likely where the issue comes from. the module is currently unfinished.

joshuarubin, 2021-07-20:
How else would I have added the rigctl server? I understand it's unfinished. I just wanted to help by offering a bug report.

Are you saying you can't reproduce _without_ the rigctl server? The bug is _with_ the rigctl server, yes, that's what I'm saying.

Are you able to reproduce _with_ the rigctl server enabled?

AlexandreRouma, 2021-07-20:
> How else would I have added the rigctl server?

All modules at the moment have to be instantiated through the config rather than the module manager (should probably disable the menu by default but it's basically there only for quick tests).

> Are you saying you can't reproduce without the rigctl server?

In this case what I think is happening is the rigctl server module relies on an event that only fires once all modules are finished loading (that way it can have an up to date list of all available VFOs and recorders). The module manager currently doesn't fire that even.

Without that event, it's not getting a list of available VFOs and therefore is using the center frequency of the IQ as a backup (intended behavior when no VFO is available).

To add the module through the config, edit the `config.json` file located either at `~/.config/sdrpp/` on linux or the root directory on windows.

You need to add an element to the `moduleInstances` parameter with the module name `rigctl_server` like so: (make sure to respect json synthax)
```
"Rigctl Server": {
    "enabled": true,
    "module": "rigctl_server"
},
```

In the latest commit the module is enabled automatically but this doesn't apply for users with existing configs.

> Are you able to reproduce with the rigctl server enabled

When enabled through the config I can't, when enabled through the module manager I can.

Regards,
Akex

joshuarubin, 2021-07-21:
Thank you for taking the time to write such a thorough response. I really appreciate it. I can confirm that enabling rigctl server from the config file does in fact resolve this issue.

New potential bug though, I'll make a new issue if you can confirm. I'm not 100% certain, but I think that if rigctl is listening before the radio starts, clicking the "play" button causes a segfault.

AlexandreRouma, 2021-07-26:
Hello,
Any updates on this bug?

joshuarubin, 2021-07-26:
It was pretty unstable for me, so I've put things aside for a bit until the rest of the app got more stable. I can try again today.

closed 2021-08-17 by AlexandreRouma