AlexandreRouma/SDRPlusPlus #178: ARM32 build and install succesful: [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig

SebNik1, 2021-07-20:
[2021-07-20 16:38:21.105] [info] SDR++ v0.3.0_beta
[2021-07-20 16:38:21.106] [warning] Root directory ../root_dev/ does not exist, creating it
[2021-07-20 16:38:21.106] [info] Loading config
[2021-07-20 16:38:21.106] [warning] Config file '../root_dev//config.json' does not exist, creating it
[2021-07-20 16:38:21.107] [warning] ConfigManager locked, waiting...
[2021-07-20 16:38:22.057] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
sdrpp: /home/pi/SDRPlusPlus/core/src/dsp/block.h:43: void dsp::generic_block<BLOCK>::stop() [with BLOCK = dsp::Reshaper<dsp::complex_t>]: Assertion `_block_init' failed.
Abgebrochen

First thank you all for your effort, SDRPlusPlus looks very promising. Yet I did not get it to run from the MacOS Installer (runs successfully, but not trace of the program afterwards) nor build from source on my Raspi4. I compiled and installed on the Raspi but I get the error above. I am not very experienced so please forgive me if it is obvious to you why it does not work. If there is anything else I should provide I will gladly try to do so. 

Thank you in advance!

My Kernel is 5.10.17-v7l+
I got the Raspbian OS:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

AlexandreRouma, 2021-07-20:
That error looks like some OpenGL issues, the raspberry pi only supports a very old OpenGL version. I don't have a pi in working order at the moment so can't find a way to get it working.

>  Yet I did not get it to run from the MacOS Installer (runs successfully, but not trace of the program afterwards)

the installer doesn't create an icon yet, the software has to be launched from a terminal by running `sdrpp`
Make sure to have all the dependencies installed as the macos package doesn't contain them.

SebNik1, 2021-07-20:
Thank you very much for the quick reply! I got the mac dependencies fixed but turns out my macos is 10.14 so I finally got my windows pc :-)

MaxHusky94, 2021-08-04:
For the Raspberry OS issue: Have a look at Tim's comment at https://www.rtl-sdr.com/sdr-version-1-0-0-released/#comment-219734 . Worked for me. I am glad.

AlexandreRouma, 2021-08-21:
Working on supporting the RPi with no code modifications necessary, please follow #260

closed 2021-08-21 by AlexandreRouma