AlexandreRouma/SDRPlusPlus #179: Set up repositories?

DC7IA, 2021-07-20:


AlexandreRouma, 2021-07-21:
Hello,

I'm already working on this but since I don't have any ubuntu or debian system I haven't finished it yet.
The SDR++ PPA is already registered.

DC7IA, 2021-07-23:
Maybe also try to get it into the Ubuntu repositories and other repositories like Debian and FreeBSD. :D 

That'd be cool!

AlexandreRouma, 2021-07-23:
> Maybe also try to get it into the Ubuntu repositories and other repositories like Debian and FreeBSD.

For this, debian requires you to be "sponsored" by a debian developper and I hate that system.
Not to mention that if it ever gets into the official repos you can be sure it'll never be updated...

For freebsd it might be possible, I know someone who messes with that.

AlexandreRouma, 2022-09-30:
Seeing how stalled it is (over a year, they're still on 1.0.3), having it on debian probably won't happen, and I'm not gonna waste time with an Ubuntu PPA because I never use ubuntu.

closed 2022-09-30 by AlexandreRouma