AlexandreRouma/SDRPlusPlus #184: Ubuntu Mate: Shows up in "other" not "hamradio"

DC7IA, 2021-07-23:
SDR++ uses the wrong category on Ubuntu Mate (and possibly other OSes with Mate Desktop?), instead of showing up in the "hamradio" category, it shows up in the category "other".

Not exactly sure if this is the right place, as I don't know what determines what category is used.

AlexandreRouma, 2021-07-23:
Hello

"hamradio" isn't a standard category. Moreover, if you would chose a non-standard category, "SDR" would make more sense.

DC7IA, 2021-07-26:
> "hamradio" isn't a standard category. Moreover, if you would chose a non-standard category, "SDR" would make more sense.

Not really, as I already have a lot of stuff in the "hamradio" category, so I don't want a new category consisting only of a single pogram.

Exhibit A:

[image]

Exhibit B:

[image]

AlexandreRouma, 2021-07-26:
Are you sure that this is an official category?

AlexandreRouma, 2021-07-26:
Alright updated the category, lemme know if it works

DC7IA, 2021-07-28:
> Are you sure that this is an official category?

It exists by default on Ubuntu, just wanted to look at FreeBSD, but my FreeBSD laptop just died when I turned it on, so cannot check right now.

closed 2021-08-19 by AlexandreRouma

shuyuan-liu, 2022-03-05:
> Are you sure that this is an official category?

I know this issue is old, but just for documentation, `HamRadio` is a registered category under Additional Categories in the freedesktop.org Desktop Menu Specification: https://specifications.freedesktop.org/menu-spec/latest/apas02.html