AlexandreRouma/SDRPlusPlus #188: Undefined symbol: rtlsdr_set_bias_tee

thedocruby, 2021-07-25:
Every time I try to use my RTL SDR, when i hit play, the program crashes and i get this:

`sdrpp: symbol lookup error: /usr/lib/sdrpp/plugins/rtl_sdr_source.so: undefined symbol: rtlsdr_set_bias_tee`

I have tried reinstalling librtlsdr0 and installing librtlsdr-dev, but to no avail. I suspect that this is a small bug in the code that will be easy to fix, just so you know I am using an RTL SDR Blog v3 with Bias T and Integrated HF Direct Sampling. I am on linux mint 21.04.

AlexandreRouma, 2021-07-25:
Make sure you're running an up to date version of librtlsdr

thedocruby, 2021-07-25:
Ah, I see, thanks! This is a really great program! An amazing Linux alternative to SDR# (i like it better anyway tbh). Great work!

P.S. A feature that I would love to see is NFM repeater uplink/downlink tone filtering so that I don't have to hear the tone in the background when I'm working a local NFM VHF or UHF repeater, and idk what you do for SDRs that transmit like the LimeSDR (which should be in my mailbox soon) but having the capability to add an uplink tone to the transmission would save me loads of trouble, otherwise I would have to route the audio from my mic through some other application to inject the uplink tone and then pass it to sdrpp somehow. That would be a pain, and having a thoughtful feature like that in the app would be fantastic and very useful.

AlexandreRouma, 2021-07-25:
SDR++ does not support TX

closed 2021-07-26 by AlexandreRouma