AlexandreRouma/SDRPlusPlus #189: macOS package miss libGLEW.2.2.dylib

effeerre75, 2021-07-25:
Installed from Actions Artifacts Build Binaries #218: Commit b327bfb on macOS Catalina.
Full error was:

dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: image not found

AlexandreRouma, 2021-07-25:
Hello,

The package doesn't have *any* dependency shipped with it, you need to install them through homebrew. They are the same as the dependencies for linux.

effeerre75, 2021-07-25:
👍👍👍 thanks I’ll try later! Macport’s user here: fingers crossed! 😃

AlexandreRouma, 2021-07-26:
Were you able to get it to work with macports? If so I'll close this issue

effeerre75, 2021-07-27:
Sorry but is not the right time to mess up with the system. I'll try ASAP! 😉

Darkcast, 2021-07-27:
I have the same isssue I installed https://formulae.brew.sh/formula/glew and I still get

dyld: Library not loaded: /usr/local/opt/glfw/lib/libglfw.3.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: image not found
[1]    32872 abort      sdrpp

aneitzel, 2021-08-09:
same here (MAC OS HighSierra), sdrpp_macos_amd64.pkg shows error:

iMac:~ alpha$ sudo sdrpp
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: image not found
Abort trap: 6

------

Alphas-iMac:~ alpha$ sudo port list glew
Warning: port definitions are more than two weeks old, consider updating them by running 'port selfupdate'.
glew                           @2.2.0          graphics/glew
Alphas-iMac:~ alpha$ 

Glew is installed truth macports before. So sad... i was happy to see that finally we have a MAC-package.
Any tips to solve that will be very appreciated. If someone is able to provide a MAC DMG-Package of SDRPP?

closed 2022-09-13 by AlexandreRouma

ea1het, 2023-02-05:
Situation remains with macOS M1 with app identifying erroneously the required architecture. CLI logs follow:

[14:55:00] jgf:~ $ sdrpp
dyld[38431]: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: <72CC01C6-4A69-39E0-BC4F-EB717CFA38CB> /usr/local/bin/sdrpp
  Reason: tried: '/usr/local/opt/glew/lib/libGLEW.2.2.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/glew/lib/libGLEW.2.2.dylib' (no such file), '/usr/local/opt/glew/lib/libGLEW.2.2.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libGLEW.2.2.dylib' (no such file), '/usr/lib/libGLEW.2.2.dylib' (no such file, not in dyld cache), '/opt/homebrew/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib' (no such file), '/opt/homebrew/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libGLEW.2.2.0.dylib' (no such file), '/usr/lib/libGLEW.2.2.0.dylib' (no such file, not in dyld cache)
[1]    38431 abort      sdrpp

michaelgeorge001, 2023-02-05:
Here is the procedure I use to compile a M1 Silicon version. I do not use MacPorts but Homebrew instead. I may have extra libraries included that are not used by SDR++. I'm sure Alexandre would be able to identify any superfluous libraries.

##SDR++ Build instructions for Apple Silicon without SDRPlay, LimeSDR, or PlutoSDR support##

```
sudo ln -s /opt/homebrew/include /usr/local/include
sudo ln -s /opt/homebrew/lib /usr/local/lib

rm -f /usr/local/bin/2to3* /usr/local/bin/idle3* /usr/local/bin/pydoc3* /usr/local/bin/python3* /usr/local/bin/python3-config*

arch -arm64 brew install airspy airspyhf cmake codec2 fftw gettext glew glfw hackrf libbladerf libusb pandoc pkg-config portaudio python3 rtaudio rtl-sdr soapysdr volk zstd

pip3 install mako

cd ~

git clone https://github.com/AlexandreRouma/SDRPlusPlus.git
cd SDRPlusPlus
mkdir build
cd build

arch -arm64 cmake .. -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_SOAPY_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_SDRPLAY_SOURCE=OFF -DOPT_BUILD_LIMESDR_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=OFF -DOPT_BUILD_PORTAUDIO_SINK=ON -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON -DOPT_BUILD_M17_DECODER=ON -DUSE_BUNDLE_DEFAULTS=ON -DCMAKE_BUILD_TYPE=Release
arch -arm64  make -j$(sysctl -n hw.logicalcpu)

cd ~/SDRPlusPlus
sh make_macos_bundle.sh ~/SDRPlusPlus/build ./SDR++.app
zip -r sdrpp_macos_silicon.zip SDR++.app

```