AlexandreRouma/SDRPlusPlus #191: Glfw Error 65543: GLX

Lesprilibre, 2021-07-27:
hi !

I have just try to intall your sdr programm on my laptop, everythnig running fine during install, but when i try to start it, i receive the following error :

Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig

I suppose that the problem is due to Mesa driver or osbolete graphic card driver ?

Lesprilibre, 2021-07-27:
I thnik i got the answer my GLEW version seems to be out of date :+1: 
---------------------------
    GLEW Extension Info
---------------------------

GLEW version 2.1.0
Reporting capabilities of display , visual 0xe4
Running on a Mesa DRI Intel(R) HD Graphics (ILK) from Intel Open Source Technology Center
OpenGL version 2.1 Mesa 20.2.6 is supported

the version should be 3.x ?

starvald, 2021-07-27:
yeah you need 3.x

Lesprilibre, 2021-07-27:
It's a laptop... Obviously, I won't update it..  😭

Lesprilibre, 2021-07-27:
I'ill give a try on the Rapi, if the version does exist, thanks !

closed 2021-07-27 by Lesprilibre

starvald, 2021-07-27:
i've ran it on the raspberry pi 4 but you'll need to self compile and alter file.

core/src/core.cpp line 278. change to 

const char* glsl_version = "#version 300 es";
glfwWindowHint(GLFW_CLIENT_API, GLFW_OPENGL_ES_API)
glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 1);

then compile.