AlexandreRouma/SDRPlusPlus #192: Program halt on soapy source

Lesprilibre, 2021-07-27:
i have a try on another platform, but when program is starting it alway freeze on this screen :

[image]

radekw8733, 2021-07-29:
Same problem here. I'm on ubuntu 21.04, sdrpp compiled on my own
[image]
[image]

radekw8733, 2021-07-29:
Update: just launched sdrpp from command line and the problem is rtl_sdr framework
[image]

radekw8733, 2021-07-29:
Again update: okay I read the faq XD
[image]

AlexandreRouma, 2021-07-29:
Hello,

Yes as mentioned in the readme, this is a common issue with soapysdr on debian based system.
Follow the instructions above to fix it.

Lesprilibre, 2021-07-29:
So, in my case I remove everything concerning soapysdr aka apt remove
soapysdr* and then reinstall just the soapy par concerning my rtlsdr
dongles ?

Thanks

radekw8733, 2021-08-02:
> So, in my case I remove everything concerning soapysdr aka apt remove soapysdr* and then reinstall just the soapy par concerning my rtlsdr dongles ? Thanks Le jeu. 29 juil. 2021 à 16:22, AlexandreRouma ***@***.***> a écrit :
> […](#)

Actually I just removed all these soapysdr packages using apt, launched sdrpp and it was everything if I remember well.

Lesprilibre, 2021-08-03:
Hi ! I think I've done nearly the same and it work on my NUC generation 8
very well !

Except that computer is generating a lot of RFI noise 🤬 and is not as well
shielded as my old Lenovo i5 X201 laptop...

But graphics libraries and drivers are not compatible on this old thing...
😭

closed 2022-08-17 by AlexandreRouma