AlexandreRouma/SDRPlusPlus #198: SoapySDR.dll is missing

travstone, 2021-08-01:
For Win64 (Win 7), with version 1.0.0, I get this error after extracting and running the exe:

The program can't start because SoapySDR.dll is missing from your computer. Try reinstalling the program to fix this problem.

I'd imagine I can just install that dll from somewhere, but thought you might want to know so you can update the install instructions/readme.

AlexandreRouma, 2021-08-01:
Hello,

It is not possible to include SoapySDR along with SDR++, it has to be installed properly on the system.

However, it is not an essential library and windows should still allow the software to start (if it doesn't, simply delete `modules/soapy_source.dll`

travstone, 2021-08-01:
Thanks for the speedy reply! I tried what you suggested, and I no longer got the error dialog. However, just as before, once the program loaded, the UI didn't look right, and behaved in a buggy way. Here's a couple of screenshots:

[image]

and after clicking around a bit to see if any of the controls are responsive:

[image]

I apologize if I missed something in the readme, but I had the impression one could just download the zip, extract, and then run the exe. If that is the intent, it seems like perhaps something isn't quite right, at least not for an older windows machine.

I'm using an RTL-SDR Blog V3 dongle, if that helps debugging at all; however, I don't even get to a step where I select the source when running SDR++. As soon as I launch the exe, I get the screens you see above.

AlexandreRouma, 2021-08-01:
Looks like your graphics drivers are broken. SDR++ makes heavy use of OpenGL.

Make sure your drivers are up to date.

closed 2021-08-17 by AlexandreRouma

fernandillos, 2022-06-21:
No logro instalar en mi HackRF ONE Portapack el SDR++ , me podría indicar como hacer para poderlo hacerlo. Gracias.