AlexandreRouma/SDRPlusPlus #199: Rewrite SDR++ in Rust for 2.0 (renaming to SDR-rs ?)

aosync, 2021-08-01:
I believe the project could benefit greatly from a rewrite in a language that is :rocket: blazing :zap: fast :bullettrain_side: and memory :computer: safe :police_officer: 
[rust](https://www.rust-lang.org/).

STBoyden, 2021-08-01:
Yes 100%, no meme

Tyrving, 2021-08-01:
I wholeheartedly agree. This needs to happen ASAP.

STBoyden, 2021-08-01:
Why use C with Classes when you can use a better language as a whole?

JackHunt, 2021-08-02:
Strong disagree. What would be the point in this particular case? What benefits would a complete rewrite of this software yield beyond the novelty of using a trendy language?

Also, @STBoyden it has been a very long time since C++ can be considered simply "C with classes".

STBoyden, 2021-08-10:
> Strong disagree. What would be the point in this particular case? What benefits would a complete rewrite of this software yield beyond the novelty of using a trendy language?
> 
> Also, @STBoyden it has been a very long time since C++ can be considered simply "C with classes".

I wouldn't worry mate, this was a joke issue created by a few people who know @AlexandreRouma.

closed 2021-08-17 by AlexandreRouma