AlexandreRouma/SDRPlusPlus #202: [RFE] Release RPMs

e-minguez, 2021-08-02:
It would be nice to have rpms to use in Fedora/CentOS/RHEL and others.

m1ctk, 2021-08-02:
Ver 1.0 exists immediately after loading ?

AlexandreRouma, 2021-08-02:
> It would be nice to have rpms to use in Fedora/CentOS/RHEL and others.

Hello,
I have no experience with these distros so help is welcome.

> Ver 1.0 exists immediately after loading ?

Please create your own issue for this.

mdomsch, 2021-09-24:
Packages for Fedora 34 and (upcoming) 35 will be in updates-testing tomorrow. 
`
$ sudo dnf --enablerepo=updates-testing install sdrpp
`

BT-Justice, 2022-06-10:
> Packages for Fedora 34 and (upcoming) 35 will be in updates-testing tomorrow.
$ sudo dnf --enablerepo=updates-testing install sdrpp

Anyway there could be a source for the nightly builds of SDR++?

BT-Justice, 2022-06-12:
Or maybe SDR++ as a Flatpak?  Flatpaks work really well.  Snaps (Ubuntu) I found to be slow.  So much so I converted from Kubuntu to Fedora KDE.

mdomsch, 2022-06-13:
@BT-Justice the Fedora RPMs have a number of patches to force use of system-provided libraries. One could use copr + webhook trigger on github commit to cause a package to be built and put into a dnf repository.  Ideally upstream would have incorporated all the patches and the spec file such that it could be built directly from a github checkout. Otherwise, you've got to somehow have and apply those patches from somewhere else during that run.

mdomsch, 2022-06-13:
@e-minguez with Fedora packages now existing and in the Fedora repositories, is this ticket ready to be closed?  Does anyone really need CentOS/RHEL packages?

e-minguez, 2022-06-13:
Ok to me to close it, thanks

AlexandreRouma, 2022-09-30:
I'll create my own RPMs because it seems the ones in the fedora package manager is missing half the damn modules.
will close when done, I suggest anyone using those distros switch the rpms I'll distribute instead

Kyr4l, 2023-10-27:
any update on this ? it's been over a year without news ?

BT-Justice, 2023-10-27:
Mine was to stop using Linux which is too fragmented for software developers to want to write great software for.  Use Windows 11.  This is what I did.  I was really missing SDR# anyway.  SDR++ works just fine on it as well.

Kyr4l, 2023-10-27:
> Mine was to stop using Linux which is too fragmented for software developers to want to write great software for. Use Windows 11. This is what I did. I was really missing SDR# anyway. SDR++ works just fine on it as well.

"Use le spyware OS"
No.

BT-Justice, 2023-10-27:
> "Use le spyware OS" No.

It appears you logged into this website.  You are being spied on by GitHub and Microsoft who owns it.  I assume you use Facebook and may X, too?  Nothing on the Internet is private.  May as well use the OS that supports all the software you like to run instead of demanding software support your OS of choice.

Kyr4l, 2023-10-27:
> > "Use le spyware OS" No.
> 
> It appears you logged into this website. You are being spied on by GitHub and Microsoft who owns it. I assume you use Facebook and X, too? Nothing on the Internet is private. May as well use the OS that supports all the software you like to run instead of demanding software support your OS of choice.

You don't know the reasons why i don't use windows besides the fact that it's spyware. Who are you to tell me which OS i should use?
Second, you're off topic, if you use windows you shouldn't even participate in this issue.

BT-Justice, 2023-10-28:
> You don't know the reasons why i don't use windows besides the fact that it's spyware. Who are you to tell me which OS i should use? Second, you're off topic, if you use windows you shouldn't even participate in this issue.

Read past comments then realize it ain't going to happen on RHEL distros unless an enthusiast takes up the task.

AlexandreRouma, 2023-10-28:
>  it's been over a year without news ?

It's been over a year without rest from university and family issues.

In any case, this convo has gotten quite offtopic, so I suggest not continuing it. 

An update will be posted when a RPM is created, and there is absolutely no ETA on it.

AlexandreRouma, 2024-10-03:
Implementation has begun in the `better_install` branch so some beta testing will soon be required on Fedora.

luzpaz, 2024-12-06:
Just checking if there's been any progress ?