AlexandreRouma/SDRPlusPlus #205: Version 1.0.0 not working

zanco48, 2021-08-02:
Not sure if this is the correct place for asking, if not than pease accept my apologies.

This morning I found out about the release version of SDR++ and decided to download and unzip it. When running the program it does start loading but eventually crashes out. In the windows event log it says the problem was caused by ucrtbase.dll . My brother also downloaded the software and with him it works just fine.

When checking ucrtbase.dll on his end is same version that I am running but has a newer date/time stamp. 

Any one have a clue how I can solve the issue? Tried several vc++ redistributable updates withou avail.

Thanks in advacne,

Jos

AlexandreRouma, 2021-08-02:
Hello,

A bit more info on the issue would be needed. What windows version are you using? What SDR were you using? What were you doing when the software crashed? etc...

zanco48, 2021-08-02:
Hi,

Ok: I am running Windows 10 64 bit 21H1 version. The RTL I am using is an RTL-SDR V3 dongle using tcp_ip.

I had no other software active and the software was starting up, showing it was initialising several things and at eht Starting Radio (or soemthing like that) it just exited the software without a message. 

Hopefully this is enough info?

AlexandreRouma, 2021-08-02:
Can you run SDR++ in a terminal (Shift + rightclick in the folder of SDR++ -> "Open powershell window here") like so:
`.\sdrpp.exe -s`

And send back a copy of everything it says from start to crash (you can select the text and right click to copy)

zanco48, 2021-08-02:
PS W:\sdrpp_windows_x64> .\sdrpp.exe -s
[2021-08-02 22:03:36.646] [info] SDR++ v1.0.0
[2021-08-02 22:03:36.651] [info] Loading config
[2021-08-02 22:03:37.583] [info] Loading icons
[2021-08-02 22:03:37.700] [info] Loading band plans
[2021-08-02 22:03:37.832] [info] Loading band plans color table
[2021-08-02 22:03:37.884] [info] Loading modules
[2021-08-02 22:03:37.931] [info] Loading ./modules/airspyhf_source.dll
[2021-08-02 22:03:38.259] [info] Loading ./modules/airspy_source.dll
[2021-08-02 22:03:38.474] [info] Loading ./modules/audio_sink.dll
[2021-08-02 22:03:38.701] [info] Loading ./modules/bladerf_source.dll
[2021-08-02 22:03:38.909] [info] Loading ./modules/discord_integration.dll
[2021-08-02 22:03:38.978] [info] Loading ./modules/file_source.dll
[2021-08-02 22:03:39.124] [info] Loading ./modules/frequency_manager.dll
[2021-08-02 22:03:39.254] [warning] ConfigManager locked, waiting...
[2021-08-02 22:03:39.254] [info] Loading ./modules/hackrf_source.dll
[2021-08-02 22:03:39.438] [info] Loading ./modules/limesdr_source.dll
[2021-08-02 22:03:39.660] [info] Loading ./modules/meteor_demodulator.dll
[2021-08-02 22:03:39.782] [info] Loading ./modules/network_sink.dll
[2021-08-02 22:03:40.173] [info] Loading ./modules/plutosdr_source.dll
[2021-08-02 22:03:40.572] [info] Loading ./modules/radio.dll
PS W:\sdrpp_windows_x64>

AlexandreRouma, 2021-08-02:
I've never seen it crash at that point.

I see you're running from a W: drive, what kind of drive is that?

AlexandreRouma, 2021-08-02:
Additionally, have you tried completely deleting your current install of SDR++ and creating a new one?

zanco48, 2021-08-03:
So the W drive is a NAS drive., So I just unzipped the files to a regular harddrive and tried to start the program using powershell. Will c/p what it said:

PS D:\sdrpp_windows_x64> .\sdrpp.exe -s
[2021-08-03 07:40:25.962] [info] SDR++ v1.0.0
[2021-08-03 07:40:25.964] [info] Loading config
[2021-08-03 07:40:25.964] [warning] Config file './config.json' does not exist, creating it
[2021-08-03 07:40:26.340] [info] Loading icons
[2021-08-03 07:40:26.439] [info] Loading band plans
[2021-08-03 07:40:26.442] [info] Loading band plans color table
[2021-08-03 07:40:26.466] [info] Loading modules
[2021-08-03 07:40:26.466] [info] Loading ./modules/airspyhf_source.dll
[2021-08-03 07:40:26.510] [warning] Config file './airspyhf_config.json' does not exist, creating it
[2021-08-03 07:40:26.512] [info] Loading ./modules/airspy_source.dll
[2021-08-03 07:40:26.533] [warning] Config file './airspy_config.json' does not exist, creating it
[2021-08-03 07:40:26.536] [info] Loading ./modules/audio_sink.dll
[2021-08-03 07:40:26.562] [warning] Config file './audio_sink_config.json' does not exist, creating it
[2021-08-03 07:40:26.564] [info] Loading ./modules/bladerf_source.dll
[2021-08-03 07:40:26.633] [warning] Config file './bladerf_config.json' does not exist, creating it
[2021-08-03 07:40:26.635] [info] Loading ./modules/discord_integration.dll
[2021-08-03 07:40:26.655] [info] Loading ./modules/file_source.dll
[2021-08-03 07:40:26.681] [warning] Config file './file_source_config.json' does not exist, creating it
[2021-08-03 07:40:26.683] [info] Loading ./modules/frequency_manager.dll
[2021-08-03 07:40:26.713] [warning] Config file './frequency_manager_config.json' does not exist, creating it
[2021-08-03 07:40:26.714] [info] Loading ./modules/hackrf_source.dll
[2021-08-03 07:40:26.735] [warning] Config file './hackrf_config.json' does not exist, creating it
[2021-08-03 07:40:26.738] [info] Loading ./modules/limesdr_source.dll
[2021-08-03 07:40:26.763] [warning] Config file './limesdr_config.json' does not exist, creating it
[2021-08-03 07:40:26.765] [info] Loading ./modules/meteor_demodulator.dll
[2021-08-03 07:40:26.768] [warning] Recordings directory does not exist, creating it
[2021-08-03 07:40:26.769] [warning] Config file './meteor_demodulator_config.json' does not exist, creating it
[2021-08-03 07:40:26.770] [info] Loading ./modules/network_sink.dll
[2021-08-03 07:40:26.794] [warning] Config file './network_sink_config.json' does not exist, creating it
[2021-08-03 07:40:26.795] [info] Loading ./modules/plutosdr_source.dll
[2021-08-03 07:40:26.847] [warning] Config file './plutosdr_source_config.json' does not exist, creating it
[2021-08-03 07:40:26.849] [info] Loading ./modules/radio.dll
[2021-08-03 07:40:26.852] [warning] Config file './radio_config.json' does not exist, creating it
[2021-08-03 07:40:26.853] [info] Loading ./modules/recorder.dll
[2021-08-03 07:40:26.878] [warning] Config file './recorder_config.json' does not exist, creating it
[2021-08-03 07:40:26.880] [info] Loading ./modules/rigctl_server.dll
[2021-08-03 07:40:26.903] [warning] Config file './rigctl_server_config.json' does not exist, creating it
[2021-08-03 07:40:26.905] [info] Loading ./modules/rtl_sdr_source.dll
[2021-08-03 07:40:26.947] [warning] Config file './rtl_sdr_config.json' does not exist, creating it
[2021-08-03 07:40:26.948] [info] Loading ./modules/rtl_tcp_source.dll
[2021-08-03 07:40:26.972] [warning] Config file './rtl_tcp_config.json' does not exist, creating it
[2021-08-03 07:40:26.974] [info] Loading ./modules/sdrplay_source.dll
[2021-08-03 07:40:27.051] [warning] Config file './sdrplay_config.json' does not exist, creating it
[2021-08-03 07:40:27.052] [info] Loading ./modules/soapy_source.dll
[2021-08-03 07:40:27.077] [error] Couldn't load ./modules/soapy_source.dll. Error code: 126
[2021-08-03 07:40:27.078] [info] Loading ./modules/spyserver_source.dll
[2021-08-03 07:40:27.137] [warning] Config file './spyserver_config.json' does not exist, creating it
[2021-08-03 07:40:27.138] [info] Initializing Airspy Source (airspy_source)
[2021-08-03 07:40:28.478] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-08-03 07:40:29.799] [info] Initializing Audio Sink (audio_sink)
[2021-08-03 07:40:29.800] [info] Initializing BladeRF Source (bladerf_source)
[2021-08-03 07:40:31.121] [error] Could not list devices
[2021-08-03 07:40:31.122] [info] Initializing File Source (file_source)
[2021-08-03 07:40:31.123] [info] Initializing Frequency Manager (frequency_manager)
[2021-08-03 07:40:31.125] [info] Initializing HackRF Source (hackrf_source)
[2021-08-03 07:40:32.459] [info] Initializing LimeSDR Source (limesdr_source)
[2021-08-03 07:40:32.460] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-08-03 07:40:32.461] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2021-08-03 07:40:33.797] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-08-03 07:40:33.799] [info] Initializing Radio (radio)
PS D:\sdrpp_windows_x64>

When I look in the windows event log it shows an error message:

Naam van toepassing met fout: sdrpp.exe, versie: 0.0.0.0, tijdstempel: 0x61069ad6
Naam van module met fout: ucrtbase.dll, versie: 10.0.19041.789, tijdstempel: 0x2bd748bf
Uitzonderingscode: 0xc0000409
Foutmarge: 0x000000000007286e
Id van proces met fout: 0x3f38
Starttijd van toepassing met fout: 0x01d7882a0f05c175
Pad naar toepassing met fout: D:\sdrpp_windows_x64\sdrpp.exe
Pad naar module met fout: C:\WINDOWS\System32\ucrtbase.dll
Rapport-id: db1f882d-f31e-442c-b8e0-6f0391b71130
Volledige pakketnaam met fout: 
Relatieve toepassings-id van pakket met fout: 

So at first run it seems to be unhappy about soapy_source.dll but when I look in the modules foldder the file is present. I hope this info is helpful, I am really unsure why it doesn't run.

hanslammerts, 2021-08-03:
FWIW, I'm having the exact same problem...

Windows10 Pro
21H1
Build 19043.1110
Windows Feature Experience Pack

AlexandreRouma, 2021-08-03:
Hmm they're very weird.

The radio module is the place I would least expect it to fail since it has no dependencies whatsoever...

Don't worry about the soapysource thing, it's just complaining you don't have soapysdr installed. it's not neccesary at all and is meant to still run without it.

I might be able to compile a special debug version to try and narrow down the issue  later on but I can't atm.

In the mean time could you give more info on the exact specs of your PC? CPU, RAM amount, GPU, etc...

hanslammerts, 2021-08-03:
Sure,

PC is a Dell laptop, I5-6200 CPU, 8 GB memory, the GPU is Intel HD Graphics 520.
I have to admit that the SDR I'm using is not connected to the laptop directly, but through VirtualHere.

My SDRPlay device (SDR1) is connected directly to a Debian server, which runs the linux server version of VirtualHere. The Windows 10 laptop runs the VirtualHere client (vhui64). This way, I can connect using usb over ip, which works with other Windows SDR clients. 
Maybe there is an issue with the usb over ip way of working, however, reading the posts of Jos (who does have his dongle connected directly, and seems to have the exact same problem) does not point in that direction.

Anyhow, glad to help if you need more info.
But beware, I know a lot more of Linux than I know of Windows.....
(as everyone should have :-) )

Regards,
Hans 
PA3CRZ

zanco48, 2021-08-04:
On my end it is an Acer TC-886, I7-9700 with 16 GB RAM. Video is NVIDEA Geforce GT710.

The SDR is installed on a seperate computer and I connect to it throught rtl_tcp which works ok with sdr#. Weird thing is also that my brother (on the same network) connects to the same server pc to a different dongle (need to check which one) but also over tcp without a problem.

So would be happy to do more testing to hopefully solve the problem

Greetings,
Jos
PE1GPE

m1ctk, 2021-08-05:
Also crashes with me just the same way and the newer 1.0.1 version with latest Windows 10 64bit build here. Something obviously needs to be looked at as it works for some not for others

zanco48, 2021-08-12:
Today tried the 1.0.3 version but still not working. Hopefully there will be a debug version to find out why the software crasshes out on start as I would love to have a go at using this software

m1ctk, 2021-08-12:
Working fine on my laptop and other pc's but not my main one with all the sdr progs on it etc. As for the mac version it doesn't even find it after sdrpp in terminal so another weird anomaly! Looking forward to a fix otherwise an excellent program with future options to decode meteor M2 it seems

LumineRose, 2021-09-10:
Upd: Problem is in Windows ver. 20H1, on earlier versions (1909 and older) there are no problems with launch

And this has nothing to do directly with the characteristics of the pc (tested on fx8300 8gb ram, gtx 960 (4gb) and intel m3-6Y30, Intel HD Graphics 515) used to be 1909 on two devices, after updating the pc to 20H1 sdr ++ stopped working (any versions and 1.0.3 ) on the tablet since 1909 everything works

log:
PS C:\Users\Ruby\Desktop\sdrpp_windows_x64> .\sdrpp.exe -s
[2021-09-11 01:12:20.427] [info] SDR++ v1.0.3
[2021-09-11 01:12:20.429] [info] Loading config
[2021-09-11 01:12:20.429] [warning] Config file './config.json' does not exist, creating it
[2021-09-11 01:12:21.060] [info] Loading icons
[2021-09-11 01:12:21.070] [info] Loading band plans
[2021-09-11 01:12:21.073] [info] Loading band plans color table
[2021-09-11 01:12:21.105] [info] Loading modules
[2021-09-11 01:12:21.106] [info] Loading ./modules/airspyhf_source.dll
[2021-09-11 01:12:21.109] [warning] Config file './airspyhf_config.json' does not exist, creating it
[2021-09-11 01:12:21.110] [info] Loading ./modules/airspy_source.dll
[2021-09-11 01:12:21.118] [warning] Config file './airspy_config.json' does not exist, creating it
[2021-09-11 01:12:21.119] [info] Loading ./modules/audio_sink.dll
[2021-09-11 01:12:21.136] [warning] Config file './audio_sink_config.json' does not exist, creating it
[2021-09-11 01:12:21.137] [info] Loading ./modules/bladerf_source.dll
[2021-09-11 01:12:21.151] [warning] Config file './bladerf_config.json' does not exist, creating it
[2021-09-11 01:12:21.152] [info] Loading ./modules/discord_integration.dll
[2021-09-11 01:12:21.167] [info] Loading ./modules/file_source.dll
[2021-09-11 01:12:21.184] [warning] Config file './file_source_config.json' does not exist, creating it
[2021-09-11 01:12:21.185] [info] Loading ./modules/frequency_manager.dll
[2021-09-11 01:12:21.201] [warning] Config file './frequency_manager_config.json' does not exist, creating it
[2021-09-11 01:12:21.202] [info] Loading ./modules/hackrf_source.dll
[2021-09-11 01:12:21.218] [warning] Config file './hackrf_config.json' does not exist, creating it
[2021-09-11 01:12:21.219] [info] Loading ./modules/limesdr_source.dll
[2021-09-11 01:12:21.235] [warning] Config file './limesdr_config.json' does not exist, creating it
[2021-09-11 01:12:21.236] [info] Loading ./modules/meteor_demodulator.dll
[2021-09-11 01:12:21.251] [warning] Recordings directory does not exist, creating it
[2021-09-11 01:12:21.252] [warning] Config file './meteor_demodulator_config.json' does not exist, creating it
[2021-09-11 01:12:21.252] [info] Loading ./modules/network_sink.dll
[2021-09-11 01:12:21.267] [warning] Config file './network_sink_config.json' does not exist, creating it
[2021-09-11 01:12:21.268] [info] Loading ./modules/plutosdr_source.dll
[2021-09-11 01:12:21.285] [warning] Config file './plutosdr_source_config.json' does not exist, creating it
[2021-09-11 01:12:21.286] [info] Loading ./modules/radio.dll
[2021-09-11 01:12:21.400] [warning] Config file './radio_config.json' does not exist, creating it
[2021-09-11 01:12:21.402] [info] Loading ./modules/recorder.dll
[2021-09-11 01:12:21.404] [warning] Config file './recorder_config.json' does not exist, creating it
[2021-09-11 01:12:21.405] [info] Loading ./modules/rigctl_server.dll
[2021-09-11 01:12:21.513] [warning] Config file './rigctl_server_config.json' does not exist, creating it
[2021-09-11 01:12:21.514] [info] Loading ./modules/rtl_sdr_source.dll
[2021-09-11 01:12:21.516] [warning] Config file './rtl_sdr_config.json' does not exist, creating it
[2021-09-11 01:12:21.519] [info] Loading ./modules/rtl_tcp_source.dll
[2021-09-11 01:12:21.610] [warning] Config file './rtl_tcp_config.json' does not exist, creating it
[2021-09-11 01:12:21.611] [info] Loading ./modules/sdrplay_source.dll
[2021-09-11 01:12:21.616] [warning] Config file './sdrplay_config.json' does not exist, creating it
[2021-09-11 01:12:21.617] [info] Loading ./modules/soapy_source.dll
[2021-09-11 01:12:21.711] [error] Couldn't load ./modules/soapy_source.dll. Error code: 126
[2021-09-11 01:12:21.711] [info] Loading ./modules/spyserver_source.dll
[2021-09-11 01:12:21.714] [warning] Config file './spyserver_config.json' does not exist, creating it
[2021-09-11 01:12:21.715] [info] Initializing Airspy Source (airspy_source)
[2021-09-11 01:12:21.863] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2021-09-11 01:12:21.981] [info] Initializing Audio Sink (audio_sink)
[2021-09-11 01:12:22.059] [info] Initializing BladeRF Source (bladerf_source)
[2021-09-11 01:12:22.087] [error] Could not list devices
[2021-09-11 01:12:22.088] [info] Initializing File Source (file_source)
[2021-09-11 01:12:22.170] [info] Initializing Frequency Manager (frequency_manager)
[2021-09-11 01:12:22.171] [info] Initializing HackRF Source (hackrf_source)
[2021-09-11 01:12:22.205] [info] Initializing LimeSDR Source (limesdr_source)
[2021-09-11 01:12:22.207] [info] Initializing PlutoSDR Source (plutosdr_source)
[2021-09-11 01:12:22.216] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2021-09-11 01:12:22.256] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2021-09-11 01:12:22.258] [info] Initializing Radio (radio)
[2021-09-11 01:12:22.388] [info] RtAudio stream open
[2021-09-11 01:12:22.389] [info] Initializing Recorder (recorder)
[2021-09-11 01:12:22.391] [info] Initializing Rigctl Server (rigctl_server)
[2021-09-11 01:12:22.392] [info] Initializing SDRplay Source (sdrplay_source)
[2021-09-11 01:12:22.400] [error] Could not intiatialized the SDRplay API. Make sure that the service is running.
[2021-09-11 01:12:22.400] [info] Initializing SoapySDR Source (soapy_source)
[2021-09-11 01:12:22.416] [error] Module 'soapy_source' doesn't exist
[2021-09-11 01:12:22.417] [info] Initializing SpyServer Source (spyserver_source)
PS C:\Users\Ruby\Desktop\sdrpp_windows_x64>

zanco48, 2021-10-20:
It is too bad that there is no follow up on this issue as far as I can see. Just downloaded and tried the 1.0.4 version and recently updated my system to Windows 11 but still crashing already at startup

AlexandreRouma, 2022-09-13:
I suggest testing the nightly and if there still is an issue, reopen this.

closed 2022-09-13 by AlexandreRouma

zanco48, 2022-09-13:
Where do I find the nightly builds? 

 

I do see a 1.0.4 version but that dates from October last year so could you point me in the right direction?

 

Thanks in advance,

 

Jos

 

Dit e-mailbericht is met AVG-antivirussoftware gecontroleerd op virussen.
www.avg.com

AlexandreRouma, 2022-09-13:
https://www.sdrpp.org/nightly