AlexandreRouma/SDRPlusPlus #210: Ver 1 : Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig

SeanMacSuibhne, 2021-08-02:
Hi
Im getting the following error while trying to run sdrpp on an up to date Linux xubuntu 20.04 Focal installation
`Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig`

if I run this command :
`glxinfo | grep version`
I get the following

```
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL version string: 2.1 Mesa 20.2.6
OpenGL shading language version string: 1.20
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 20.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16

```
Is there some thing else I should do to fix this problem?

AlexandreRouma, 2021-08-02:
Your graphics drivers are out of date.

OpenGL 3.0 is requires for SDR++

SeanMacSuibhne, 2021-08-02:
All drivers and software are up to date so maybe my hardware is too old? The laptiop is 10 years old.
Ill be sticking with GQRX so I suppose.

AlexandreRouma, 2021-08-02:
The latest version your laptop supports (2.1) is over 15 years old. I can't support this without performance loss for every other user.
Switching to OpenGL 2.1 requires switching out the entire ImGui backend, so it's not as simple as one cmake flag.

SeanMacSuibhne, 2021-08-02:
Well thank you for responding and putting me out of my misery. Im not sure I need a new laptop but maybe someday!
Keep up the good work.

Seán

alphafox02, 2021-08-03:
Would this allow it to start or is it just not even possible with 2.1? 

I had to use something like this on the Pi, except it was 4.5.  

MESA_GL_VERSION_OVERRIDE=2.1 sdrpp

SeanMacSuibhne, 2021-08-04:
Hi
Thank you again for all you work tring to get SDRPP working for me
`MESA_GL_VERSION_OVERRIDE=2.1 sdrpp` 
did not work it gave the same error message.

I had a look to see which version was installed  ahd I have mesa-utils version 8.4.0-build1 installed.
So I tried: 
`MESA_GL_VERSION_OVERRIDE=8.4.0 sdrpp`
And got this:
```
[2021-08-04 06:25:40.426] [info] SDR++ v1.0.0
[2021-08-04 06:25:40.426] [info] Loading config
Segmentation fault (core dumped)
```
It also tried to put a window up but it close it as soon as the error message appearad.

AlexandreRouma, 2021-08-04:
Could you update to the SDR++ version I published yesterday (1.0.1)?

The config is likely corrupted after so many crashes and it'll get fixed automatically by this new version

Lesprilibre, 2021-08-08:
I did have same message concerning GL driver on my old laptop (Lenovo X201), so I decide to give a try on a VM machine on this same laptop using virtualbox.

On a fresh Debian minimal Debian install I have installed dependencies and the last .Deb of SDR++ for burster and it works !

ol2tmx, 2021-08-11:
MESA_GL_VERSION_OVERRIDE=3.3 sdrpp is starting up now but does not decode any stream with SoapySDR

AlexandreRouma, 2021-08-21:
moving this issue to #260 
Please follow that one for updates on older GL support

closed 2021-08-21 by AlexandreRouma