AlexandreRouma/SDRPlusPlus #212: [Feature]Spectrum averaging

Radiomix2000, 2021-08-02:
Thank you for your software! It becomes more and more awesome.
Can I ask, how to apply averaging to spectrum plot?
If there is no such feature, is any chances to see it in future releases?

AlexandreRouma, 2021-08-02:
Hello,

This is a planned feature is denoted in https://github.com/AlexandreRouma/SDRPlusPlus/projects/2

0xCoto, 2021-09-14:
Any updates on this? Definitely a must-have in my opinion! 🙂

AlexandreRouma, 2021-09-14:
it's in the todo list but I've been more focused on bugfix lately

AlexandreRouma, 2023-03-30:
has been implemented. See "FFT Smoothing" and "FFT Smoothing Speed" options in the "Display" menu.

closed 2023-03-30 by AlexandreRouma

helioshk, 2024-09-27:
Can the FFT smoothing also be applied to the waterfall?

KubaPro010, 2024-09-27:
@helioshk No, there's no need for waterfall smoothing at all

AlexandreRouma, 2024-09-27:
Feel free to smear some vaseline on your screen if you'd like smoothing on the waterfall. It would be a nonsensical feature...

helioshk, 2024-09-27:
It would increase the visual SNR for persistant signals. (Average of noise gets lower while the average of the signal stays the same)

AlexandreRouma, 2024-09-27:
Just increase the FFT size and decrease the FFT rate. It'll integrate over a long time

helioshk, 2024-09-27:
It doesn't seem to go below 1/s

helioshk, 2024-09-28:
I think it isn't integrating at all. Or am I doing something wrong?
I see no difference between FFT framerate of 1 vs 30, apart from all levels being a bit lower. But the visual SNR stays the same:
[image]