AlexandreRouma/SDRPlusPlus #217: didn't work on Mac

cn0xroot, 2021-08-03:
```
sdrpp
dyld: Library not loaded: /usr/local/opt/glfw/lib/libglfw.3.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: image not found
[1]    70951 abort      sdrpp
```

install via https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.0/sdrpp_macos_amd64.pkg

osx 11.4

cn0xroot, 2021-08-03:
already install glfw with homebrew

aneitzel, 2021-08-10:
same here... installed glfw via macports before but SDRpp dont find lib at start. A step by step guide how to get it run under MacOS will be good.

AlexandreRouma, 2021-08-10:
Issue seems to be macport isntalling packages where the dynamic linker can't find them. Either configure macos's dynamic linker to find the packages (I have no idea how or if it's even possible, I don't use osx myself) or just use homebrew.

m1ctk, 2021-08-11:
It installed ok but I cant find it anywhere !

xologram, 2021-08-11:
same, installed but nowhere to be found. on m1 mac but figured should work via rosetta?

AlexandreRouma, 2021-08-11:
it does not create an icon, run it via the terminal by running `sdrpp`.

m1ctk, 2021-08-11:
> it does not create an icon, run it via the terminal by running `sdrpp`.

Doesn't work with Big Sur 11.5.1

xologram, 2021-08-11:
thanks @AlexandreRouma ! 

I too get the same error as OP:

`$ sdrpp
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: image not found
Abort trap: 6`

since homebrow installs libGLEW in different location had to symlink the lib:

`mkdir -p /usr/local/opt/glew/lib/
sudo ln -s /opt/homebrew/lib/libGLEW.2.2.dylib /usr/local/opt/glew/lib/libGLEW.2.2.dylib` 

but now when I try to start it I get wrong arch :( 

`$ sdrpp
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: no suitable image found.  Did find:
        /usr/local/opt/glew/lib/libGLEW.2.2.dylib: mach-o, but wrong architecture
        /opt/homebrew/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib: mach-o, but wrong architecture
Abort trap: 6`

no luck on amd64 arch

PiTiLeZarD, 2021-08-13:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/244#issuecomment-898230736

Should've read here first haha

aosync, 2022-08-17:
Instructions on how to build on MacOS have been added by #402. Feel free to reopen if the issue persists.

closed 2022-08-17 by aosync