AlexandreRouma/SDRPlusPlus #227: crashing on load & bladerf not detected ver 1.0.1

pmmeasures-tech, 2021-08-06:
Ver 1.0.1 on windows 10 21H1
On install - app crashed out, well more collapsed with no error message.

It was at the Recording Module load bit - managed to get passed by deleting the recorder_config.json

Program does not see my Bladerf :-( managed to get working by entry add down in module manager panel and a close and reopen of program.

Swapping device to airspy - does not detect only picks up when closing and restarting program

Seems unless you have all your devices connected at program start you wont have access when plugging in after program started.

AlexandreRouma, 2021-08-06:
Hello,

The only way this type of crash could happen is if you updated from one of the beta versions.
same thing for the BladeRF module not being loaded.

As for the airspy not being picked up, did you press the "refresh" button?

pmmeasures-tech, 2021-08-07:
Arhhhh ok that makes sense as had an earlier version 👍 will try afresh :-)

closed 2021-08-17 by AlexandreRouma