AlexandreRouma/SDRPlusPlus #228: v1.0.1 Fatal error Module 'sdrplay_source' doesn't exist on Ubuntu 21.04

peterbmarks, 2021-08-07:
A lovely looking splash screen but I get a fatal error before the app starts:

[2021-08-07 14:34:19.729] [info] RtAudio stream open
[2021-08-07 14:34:19.729] [info] Initializing Recorder (recorder)
[2021-08-07 14:34:19.730] [info] Initializing Rigctl Server (rigctl_server)
[2021-08-07 14:34:19.731] [info] Initializing SDRplay Source (sdrplay_source)
[2021-08-07 14:34:19.732] [error] Module 'sdrplay_source' doesn't exist
[2021-08-07 14:34:19.732] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4build1
Hash collision!!! Fatal error!!
^C

I have all the required libraries installed.

Peter

AlexandreRouma, 2021-08-07:
Hello,

As mentioned in the [troubleshooting section of the readme](https://github.com/AlexandreRouma/SDRPlusPlus#hash-collision-error-when-starting), this is a soapy SDR bug in debian based distros when installing the `soapysdr-module-all`. Please uninstall this package and install only the core of soapysdr and the modules you need to use it with.

peterbmarks, 2021-08-07:
Thankyou very much. Sorry to waste time.

closed 2021-08-07 by AlexandreRouma

peterbmarks, 2021-08-07:
I did have the module installed but I have now removed it:
marksp@optiplex:~$ sudo apt remove soapysdr-module-all
[sudo] password for marksp: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  soapysdr-module-all
0 to upgrade, 0 to newly install, 1 to remove and 0 not to upgrade.
After this operation, 6,144 B disk space will be freed.
Do you want to continue? [Y/n] 
(Reading database ... 232576 files and directories currently installed.)
Removing soapysdr-module-all:amd64 (0.7.2-1build1) ...
....
[2021-08-08 09:08:01.123] [info] Initializing Recorder (recorder)
[2021-08-08 09:08:01.124] [info] Initializing Rigctl Server (rigctl_server)
[2021-08-08 09:08:01.125] [info] Initializing SDRplay Source (sdrplay_source)
[2021-08-08 09:08:01.126] [error] Module 'sdrplay_source' doesn't exist
[2021-08-08 09:08:01.126] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4build1
Hash collision!!! Fatal error!!
^C

I'll keep looking and report back.
Thanks again for this wonderful project.
Peter

peterbmarks, 2021-08-07:
Solved now. I removed everything soapy with:
sudo apt remove 'soapysdr*'

Now I'm up and running.