AlexandreRouma/SDRPlusPlus #229: SDR++ Virtual Machine.  Will not start

M0CRT, 2021-08-07:
Hi

What a fantastic program.  Works great on my Ubuntu VM (Unraid), but not on my Windows VM (10 latest feature pack).

Extracted to My Documents folder, run normally or 'with Administrator perms...nothing.  Initially it did open a dos box momentarily but subsequent attempts...nothing.  Are there any logs I can query?

Latest 1.0.1 release for X64.

Keep up the fantastic work.

AlexandreRouma, 2021-08-07:
Hello,

SDR++ needs OpenGL and your VM likely doesn't have that.
Only way i see this working is to install an OpenGL software rendered but I haven't done that myself.

m1ctk, 2021-08-11:
> Hello,
> 
> SDR++ needs OpenGL and your VM likely doesn't have that.
> Only way i see this working is to install an OpenGL software rendered but I haven't done that myself.

Would this apply to a non VM environment as with me it crashes each time after  'initializing radio" ?

AlexandreRouma, 2021-08-11:
It would but it wouldn't get all the way to that point if it didn't have OpenGL.

I think there's something else going on.

lwvmobile, 2021-11-16:
Works fine in Virtualbox in a VM, use it in both Windows 7 and LM Virtualbox. Need to enable Enable 3D Acceleration in the Display settings for the VM.