AlexandreRouma/SDRPlusPlus #233: Testing V 1.0.1 troubles with RTL-TCP, cross modulation

Lesprilibre, 2021-08-08:
First, very happy to find a new software to replace the old but good GQRX, with a lot of more options and a greatly impproved UI !
Nice job !

By the way i'm currently testing the Aero VHF band, and i have : 
- crossmodulation from distant channels (1 MHz) on my external dongle RTL SDR V3 with a dipole antenna and Pi w/ rtl-tcp
- no possible to reduce receving bandwith under 6 kHz 
- work fine with my internal dongle on USB Nooelec NESR V4 !

Frequency control works great with Gpredic ! ;-)

AlexandreRouma, 2021-08-08:
Hello,

What sample rate are you using? 250KHz on the RTL-SDR WILL cause images.

Lesprilibre, 2021-08-08:
I finally remove TRL_TCP to replace it with by **SpyServer**, it seems to work better, but i have also change the USB of the dongle, as i remember on raspberry Pi 3B+ wifi is sharing some ports with one of the USB controler.

On the local network SampleRate at 2.4 MHz workfine for the moment, i thinks this should work fine on the internet as i have a upliknk up to 500 Mbits.....

if you want to give it a try i can sand you my public ip, and setup my routeur....

Lesprilibre, 2021-08-08:
It seems i have got an  issue with ALSA on my Mint Distro, all have been corrected and run smoothly now !

Lesprilibre, 2021-08-08:
Get back to RTL-TCP cause GQRX dosen't support spyserver, works well on WiFi and my new setup on the RaPi part...

closed 2022-08-17 by AlexandreRouma