AlexandreRouma/SDRPlusPlus #234: Documentation updates:  Mac, FreeBSD build-from-source

pcf000, 2021-08-08:
I built sdr++ successfully from source on my Big Sur Mac -- I'm listening to local FM radio right now.  I swear it's better quality sound than other rtl-sdr projects I've tried.

I followed the Linux/BSD build instructions and used brew for the package dependences.  Some of them have different names, thus

    brew install librtlsdr soapysdr soapyrtlsdr cmake fftw glfw glew volk rt-audio

I needed to disable airspy, airspyhf, hackrf, and plutosdr modules.

I also built sdr++ successfully on FreeBSD on Raspberry Pi 4, but haven't managed to run it yet;  openGL doesn't seem to like a VNC connection.  Again, some dependences have different names, thus

    pkg install cmake fftw3 glfw glew volk SoapySDR SoapyRTLSDR libiio libad9361-iio rtaudio

I also installed mesa-dri and libosmesa while trying to get past the openGL errors, but I don't know if those are strictly necessary.

FWIW, both these systems use clang, which emits a lot of compiler warnings while building.

joe-azzara, 2021-08-10:
I had brew install librtlsdr soapysdr soapyrtlsdr cmake fftw glfw glew volk rt-audio
on mac os Big Sur, it went through but nothing else happens, then went to installed sdr++ from the releases page still nothing happened can anyone help???

pcf000, 2021-08-11:
Assuming that the window at least comes up for you, make sure to set your source and sink -- on mine, they're RTL-SDR and Built-In Output, and at least the source defaulted to something like File instead.

joe-azzara, 2021-08-11:
I had gone on discord and found out the problem. Thanks

aneitzel, 2021-08-19:
can you write a litle guide here whats needed to get it run under MacOS? Pleeeeeeeeeese..... like cloning from git to local drive and then?

MCKLtech, 2021-08-22:
@pcf000 , how did you "disable airspy, airspyhf, hackrf, and plutosdr modules."?

I'm getting an error of dyld: Symbol not found: _airspy_list_devices which I assume is related to this.

AlexandreRouma, 2021-08-22:
this error happens you have have an outdated version of libairspy or libairspyhf.

To disable modules, you can use the option names listed in the module list in the readme. For example, the airspy_source module uses the OPT_BUILD_AIRSPY_SOURCE parameter. To disable it, add the following argument to the cmake command: `-DOPT_BUILD_AIRSPY_SOURCE=OFF`

MCKLtech, 2021-08-22:
Thanks @AlexandreRouma. I'm using the MacOS release with the aim of adding to the docs, but I may download the source and compile from there.

AlexandreRouma, 2021-08-22:
In that case, simply install a more up to date version of libairspy and the error should go away

MCKLtech, 2021-08-22:
Thanks again. In case anyone comes across this issue, this was my fix:

> brew install airspy

pcf000, 2021-08-22:
I built with

>     cmake .. -DOPT_BUILD_AIRSPY_SOURCE=0 -DOPT_BUILD_AIRSPYHF_SOURCE=0 -DOPT_BUILD_HACKRF_SOURCE=0 -DOPT_BUILD_PLUTOSDR_SOURCE=0

to turn off everything that caused me trouble.

(It might be useful to include an example of disabling a module in the documentation, because it wasn't immediately obvious.)

pcf000, 2021-08-22:
Sorry for the delay, but here's what I did, from a clean start on macOS 12 beta:

>     git clone git@github.com:AlexandreRouma/SDRPlusPlus.git       # ssh mode, for preference
>     # See "Building on Linux / BSD."  SoapySDR is necessary, and I have an RTL-SDR so librtlsdr.
>     # I don't know what plutosdr or airspy are, and/or couldn't find suitable versions.
>     # Some dependences have different names in homebrew.
>     brew install librtlsdr soapysdr soapyrtlsdr cmake fftw glfw glew volk rt-audio
>     cd SDRPlusPlus
>     mkdir build
>     cd build
>     # Use -D...=0 to disable the parts I don't have.
>     cmake .. -DOPT_BUILD_AIRSPY_SOURCE=0 -DOPT_BUILD_AIRSPYHF_SOURCE=0 -DOPT_BUILD_HACKRF_SOURCE=0 -DOPT_BUILD_PLUTOSDR_SOURCE=0
>     make -j<N>
>     cd ..
>     sh ./create_root.sh
>     cd build
>     sudo make install

Building on FreeBSD was nearly the same, with slightly different (again) dependency package names.

closed 2022-09-30 by AlexandreRouma