AlexandreRouma/SDRPlusPlus #235: libGLEW missing from Ubuntu Focal release and not installed with the dependencies

waterwin, 2021-08-08:
$ sudo sdrpp
sdrpp: error while loading shared libraries: libGLEW.so.2.1: cannot open shared object file: No such file or directory
but: Setting up libglew-dev:amd64 (2.0.0-5) ... worked

What have I done wrong or forgotten about?

de Erwin
PE3ES / F4VTQ

AlexandreRouma, 2021-08-08:
Hello,

Yes as mentioned in the readme, the deb doesn't currently contain the depdencies.

By the way, I see you're running SDR++ as root, I wouldn't recommend this.

waterwin, 2021-08-08:
Indeed, that is why I installed all dependencies first as listed in the README, but it looks as if this dep is not listed ...That is why I also copied the line where another GLEW related dep was shown as being installedI tried running with sudo as running without failed.
How can I install this missing dependency or are there more not listed?

STBoyden, 2021-08-10:
libglew-dev is listed as a dependency [here](https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/readme.md#linux) in the formatted code block, it's the second in the list of dependencies. It's possible you may have missed it when reading the instructions.

waterwin, 2021-08-11:
See my original comment please :

but: Setting up libglew-dev:amd64 (2.0.0-5) ... worked

So it was installed with the other dependencies and it looks as if I did not miss it, still the installed SDR++ complains

What else can I do to test my install?

AlexandreRouma, 2021-08-11:
I don't know, there's some weird stuff going on with your system.

you shouldn't have to specific `:amd64` at the end unless you already have a 32bit version of the lib for some reason.

I've never used a distro that mixes 32 and 64bit so no idea how that sort of thing works

waterwin, 2021-08-11:
That is not something I added, it is a copy of the sudo apt result from the Terminal window when getting all dependencies installed based on the information for SDR++
I did not specify anything at all.
I will do a new built on a clean Ubuntu 1804 system and get back.

waterwin, 2021-08-11:
New install on clean and updated 20.04.2 Ubuntu. But as it is a HP EliteBook 2540p I am now here : Glfw Error 65543
Read the issue on openGL versions. So I think this laptop is not going to do the work so it is back to my VM then.

waterwin, 2022-03-03:
I am revisiting the Ubuntu 18.04 install with the libGlew issue

Checked:
erwin@UBU-1804-1:/usr/bin$ glxinfo |grep "OpenGL version"
OpenGL version string: 3.1 Mesa 20.0.8

So it has the right openGL and the pc is more than capable of running it, it is in a Oracle VM VirtualBox

closed 2022-03-03 by waterwin