AlexandreRouma/SDRPlusPlus #237: LimeSdr does not work correctly (no signal)

netdoggy, 2021-08-09:
I haven't been able to quickly find exactly what the problem is (technically). However, the waterfall looks like a gain of 0 and does not change in any way when you change the gain or antenna channel.

AlexandreRouma, 2021-08-09:
1) is the input set to LNAW
2) Raise the min value

A screenshot would be useful to see what the error is.

netdoggy, 2021-08-09:
While I was taking screenshots, I did some research and it turned out that it is the Ch2 channel that does not work. For technical reasons I use the default CH2 for reception.

I switched to CH1 and it looks like a signal appeared. So the problem is probably in the choice of the antenna channel.

netdoggy, 2021-08-09:
[image]
[image]

netdoggy, 2021-08-09:
here :) 
https://github.com/AlexandreRouma/SDRPlusPlus/blob/fa9347f2eed756c31e339291d343526f069e5186/limesdr_source/src/main.cpp#L338

AlexandreRouma, 2021-08-09:
I can change that but I have no way to test it (I only have a LimeSDR mini)

Can you check that `0` switching to `_this->chanId` fixes the issue?

netdoggy, 2021-08-09:
the problem is probably here https://github.com/AlexandreRouma/SDRPlusPlus/issues/237#issuecomment-895553624

netdoggy, 2021-08-09:
Unfortunately, I'm not ready now build it, but I'll check how the fix will be :)

AlexandreRouma, 2021-08-09:
Fixed it in the latest 1.0.3 release

netdoggy, 2021-08-10:
Cool, it works, thanks.

closed 2021-08-10 by AlexandreRouma