AlexandreRouma/SDRPlusPlus #240: sdrpp won´t start - won´t pass "Initializing SoapySDR Source" on Ubuntu 21.04

chexclaim, 2021-08-10:
Hello,  I am trying to run SDR++ on Ubuntu 21,04. Installation went no problems but when I run the app it won´t pass the "Initializing SoapySDR Source" stage. And Ubuntu keeps asking if I would like to wait or force quit.
Thank you.

AlexandreRouma, 2021-08-10:
Hello,
this is likely related to [this issue](https://github.com/AlexandreRouma/SDRPlusPlus#hash-collision-error-when-starting).

chexclaim, 2021-08-10:
Yes! I ran:
apt remove soapysdr*

and the it started fine.  I DIDN´T do anything else. I did not install anything else but seems to work. 
My hardware is: Realtek RTL2832 SDR

Thank you.

AlexandreRouma, 2021-08-10:
Yes that's not an issue, only install soapysdr if you actually need it.

SDR++ has dedicated support for the RTL-SDR so you won't need soapy.

closed 2021-08-10 by AlexandreRouma