AlexandreRouma/SDRPlusPlus #244: Doesn't seem to be working on Big Sur 11.5.1.

afpena, 2021-08-12:
It does install, and I can see it listed on usr/local/bin/, but whenever you try to run it... 

XXXXXX@yyyyyy ~ % sdrpp     
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: image not found
zsh: abort      sdrpp

Sudoing doesn't do the trick either.

PiTiLeZarD, 2021-08-12:
Yup, same problem here on Big Sur (M1)

loewal, 2021-08-13:
Worked fine for me under Big Sur and even now under Monterrey.
But I must say that I used homebrew to install glew.

PiTiLeZarD, 2021-08-13:
Aha yeah so:

```
brew install glew
sudo mkdir -p /usr/local/opt/glew/lib/
sudo ln -s /opt/homebrew//lib/libGLEW.2.2.dylib /usr/local/opt/glew/lib/libGLEW.2.2.dylib
```

would get you there. But on the M1 I now have 

```
dyld: Library not loaded: /usr/local/opt/glew/lib/libGLEW.2.2.dylib
  Referenced from: /usr/local/bin/sdrpp
  Reason: no suitable image found.  Did find:
	/usr/local/opt/glew/lib/libGLEW.2.2.dylib: mach-o, but wrong architecture
	/opt/homebrew/Cellar/glew/2.2.0_1/lib/libGLEW.2.2.0.dylib: mach-o, but wrong architecture
[1]    84238 abort      sdrpp
```

Which meansI have to suss out how to install glew with brew using rosetta!

PiTiLeZarD, 2021-08-13:
Might be worth adding this to the doc? I don't know who maintains this, I can help if needed (for the doc, I'm not a C/C++ guy unfortunately ;))

xologram, 2021-08-13:
can confirm worked for me with "ibrew" method and parallel x86 homebrew!

afpena, 2021-08-13:
Worked after going through the entire glew installation. Thank you all for the support!

AlexandreRouma, 2022-09-13:
Likely fixed in the nightly but still no way to test. reopen if not.

closed 2022-09-13 by AlexandreRouma