AlexandreRouma/SDRPlusPlus #245: Started one time and then no more - Ubuntu 20.04 Pluto+

waterwin, 2021-08-12:
Ubuntu 20.04
SDR++ built from source
Pluto+

Works 1 time on FM with good audio quality, but now:
[2021-08-12 23:08:05.750] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104501000.0!
[2021-08-12 23:08:06.228] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104500000.0!
[2021-08-12 23:08:21.034] [error] Could not open pluto
[2021-08-12 23:08:21.034] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 104500000.0!

Have a complete copy of the terminal messages when needed

de Erwin
PE3ES / F4VTQ

waterwin, 2021-08-12:
[image]

AlexandreRouma, 2021-08-12:
never seen that issue with the real pluto and I don't have a pluto+.

I'd suggest contacting the manufacturers of the pluto+ since the issue is coming from inside libiio

waterwin, 2021-08-13:
No difference with Adalm Pluto, and Pluto+ now works but Adalm Pluto does not work consistently

>>Pluto+ issue comes from libiio
verify libiio version / moving target due to changes at Analog Devices
Today no issues with Pluto+, only seeing some delay/short freeze when moving Min slider at right hand side of screen
test Adalm Pluto, works with ip:pluto.local started from icon
first it works then in FM/WFM same crash when using sliders at right side of screen
restart sdrpp -s
[2021-08-13 10:13:59.282] [info] Ready.
ERROR: Avahi Resolver: Failed to resolve host 'pluto.local' : Timeout reached
ERROR: Avahi Resolver: Failed to resolve host 'pluto.local' : Timeout reached
[2021-08-13 10:14:28.369] [error] Could not open pluto
[2021-08-13 10:14:28.369] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 94906250.0!
not finding 192.168.2.1 either

will redo same test after reboot later today

waterwin, 2021-08-13:
Looks a bit more stable with the occasional hick up (I did not see differences between Adalm Pluto Rev.B or Pluto+ Rev.C) :
6. sdrpp -s all of a sudden [2021-08-13 22:13:28.808] [info] PlutoSDRSourceModule 'PlutoSDR Source': Stop!
SDR++ window still open but can be closed 
reopen [2021-08-13 22:13:28.808] [info] PlutoSDRSourceModule 'PlutoSDR Source': Stop!
dis- and re-attach Pluto+ to VM, works again,
clicking in the waterval is completely off from clicking in spectrum when using bullseye
closes [2021-08-13 22:35:01.831] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 92702905.57421875!
ERROR: READ LINE: -104
When moving Manual PGA gain slider
Pluto+ needs to be reconnected via usb cable to pc
Again PGA Gain 76 Moving in waterfall
[2021-08-13 22:38:25.758] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 92702905.57421875!
ERROR: READ LINE: -32
ERROR: READ INTEGER: -32
this continues till CTRL-C
dis- and re-attach Pluto+ to VM, works again
Pluto closes on setting of VFO; 5 MHz BW
[2021-08-13 22:58:00.560] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 5091975655.15!
ERROR: READ LINE: -32
ERROR: READ INTEGER: -32
etc
dis- and re-attach Pluto+ to VM, works again
[2021-08-13 22:58:00.560] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 5091975655.15!
[2021-08-13 22:59:58.570] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 5091975655.15!
ERROR: READ LINE: -32
ERROR: Open unlocked: -32
[2021-08-13 23:00:03.654] [error] Could not create RX buffer
Again
[2021-08-13 23:03:07.195] [info] PlutoSDRSourceModule 'PlutoSDR Source': Tune: 6084654250.0!
ERROR: READ LINE: -32
ERROR: READ INTEGER: -32
CTRL-C

waterwin, 2021-08-13:
On the IP address, yes I need to differentiate the Adalm from the Plus, so both have non default IP address 8 and 11

AlexandreRouma, 2022-09-13:
probably fixed in more recent versions, reopen if not

closed 2022-09-13 by AlexandreRouma