AlexandreRouma/SDRPlusPlus #248: Audio cut out MacOS 11.4

chrjohns, 2021-08-13:
Very grateful to have this app working natively on the Mac.  Beautiful audio when SDRPP is the frontmost window, but audio cuts out momentarily and intermittently when transitioning to another app or window or when other operations unrelated to SDRPP take resources.  It settles back to smooth sound pretty quickly.

AlexandreRouma, 2021-08-13:
which build are you using? built yourself or an official release?

chrjohns, 2021-08-13:
1.03 - you helped me build it in discord (northcountry)

AlexandreRouma, 2021-08-14:
I see,
I'll have to try it out again, don't have access to a mac anymore for now

chrjohns, 2021-08-14:
Happy to help test things if you need it.

AlexandreRouma, 2022-09-13:
Likely fixed in the nightly, please reopen if not.

closed 2022-09-13 by AlexandreRouma