AlexandreRouma/SDRPlusPlus #251: Not so Cross-Platform...AT ALL!

adamsj182, 2021-08-14:
Mr. Rouma,
Pardon me for saying this, but your software does not work on any platforms that I have. After trying over and over on Win7, MacOS 10.13, and two flavors of brand new Linux installations on both AMD64 as well as ARM64, I completely give up, which is very unusual for me. All things considered, it looks real pretty and cool and fun to play with, but if I have to go back and reverse engineer the entire program or install a million-and-one obscure dependencies that force me to upgrade my operating systems, then I've really no use for your efforts.

When you make it backwards compatible to Win7 and MacOS 10.13 and consequently making it truly a cross-platform software then let me know. As long as it only works with Win8.1+ and MacOS 11+ aside from not working on any Linux variety then it is not of any use to me and the half-dozen or so SDR programs that actually work with minimal effort on all the aforesaid platforms offering all the same features but maybe not as pretty should serve me quite well for the time being.

Anyway, good day sir.

shuyuan-liu, 2021-08-14:
Hi @adamsj182,

If you find "the half-dozen or so SDR programs that actually work... offering all the same features" to be easier to use, you can of course stick to them. No one is expecting you or anyone else to use SDR++. If you don't mind though, I believe people would always be happy to help with compiling and/or installing SDR++ if provided enough information on the exact problems you are encountering.

AlexandreRouma, 2021-08-14:
Hello,

> After trying over and over on Win7
Make sure your graphics drivers are up to date and you have vc-redistributable 2019 installed (normally needed by most modern software). Any PC made in the last 10 years will run SDR++ just fine...

> MacOS 10.13

This version is too old to be supported as apple took years to support modern C++ features. Update to at least 10.15

> and two flavors of brand new Linux installations on both AMD64 as well as ARM64

More info on your issue would be needed. But as mentioned on the release page, the **four** dependencies are listed in the readme. that's not "a million and one". The ones for OSX are the same as the ones for linux.

My software is cross platform, not cross decade.

Regards,
Alex

closed 2021-08-17 by AlexandreRouma