AlexandreRouma/SDRPlusPlus #252: Feature wish - homebrew cask for mac installation

mikkosaarisalo, 2021-08-14:
Please provide a homebrew cask for mac installation.
https://brew.sh/

jkrabbenbos, 2021-08-15:
I agree, as you already have to install the whole libglew environment through homebrew. If I download a .pkg file I expect that everything is included. 

I'm not sure if it is possible to create a cask that contains all the dependencies. I guess it will be possible.

darksidelemm, 2021-08-15:
I would suggest that producing an app-bundle with all the dependencies included is probably a more useful option? It's the 'usual' way for OSX, and has been done for other ham radio applications (CubicSDR, FreeDV, fldigi)

AlexandreRouma, 2021-08-15:
I've got a friend who started working on a standard DMG, but he's busy at the moment. Without having a Mac myself it's not easy to do (can't really run it in a VM) and I sure don't want to spend any money towards anything from apple...

So I'll have to see what he comes up with in the next few weeks.

darksidelemm, 2021-08-15:
Yep, without access to a mac it's not easy to do! (You'd be experimenting blindly with github actions I guess...) 
The other point to note is that you'll most likely have to make two app bundles - one for intel machines, and one for apple silicon. 
When building, its also possible to set min osx version cflags like here: https://github.com/drowe67/freedv-gui/blob/master/build_osx.sh#L22
This makes sure that those on older OSX versions still have a chance to use the software.

AlexandreRouma, 2021-08-15:
I won't be able to do any apple silicon binary version unfortunately. At least not until github supports it in Actions.
AS for old OSX versions, I might be able to support them in the future, not sure yet, I'll make an announcement about it if I do.

darksidelemm, 2021-08-15:
Yes, you eventually hit a wall at how old you can support due to libraries, however an artificial limitation does come from compiler options - by default when building an app bundle often it'll just default to only supporting the OSX version it was built on, when in reality the same app-bundle could be used on much older versions.

AlexandreRouma, 2022-09-13:
Since I have no mac I will not be able to take on the responsibility of this.

closed 2022-09-13 by AlexandreRouma