AlexandreRouma/SDRPlusPlus #257: Win10 - vcruntime140_1.dll not found - sdrpp not starting

waterwin, 2021-08-14:
Code execution can not proceed..

What should be my next step to get it started on Win10?

Thank you,

de Erwin
PE3ES / F4VTQ

AlexandreRouma, 2021-08-14:
You need to install [Visual C++ redistributables 2019](https://aka.ms/vs/16/release/vc_redist.x64.exe). This is normally on all windows 10 PCs but yours doesn't seem to have it.

jkrabbenbos, 2021-08-15:
@AlexandreRouma as far as I know, the redistributables are not by default on a Windows machine. However if you have software running that is using the correct redistributables, there is a chance that they are installed by that program. If you have a Windows development environment (Visual Studio) they are installed. 
My suggestion: add a comment to the information to the Windows install page and a link to the download page.

waterwin, 2021-08-15:
As can be seen from comments on the Internet : Most programs will come preinstalled with a copy of the Visual C++ Redistributable during installation. You can also browse the application setup files – normally a folder called “third party” or something similar – where you might be able to find copies of the setup files for the Visual C++ Redistributable.
The application normally needs to deliver the correct file, it is not installed on every Win10 installation as standard.
It can be found here : https://www.microsoft.com/en-us/download/details.aspx?id=52685 for the 140.dll (MSVCR140.dll (2015))

closed 2022-08-17 by AlexandreRouma