AlexandreRouma/SDRPlusPlus #258: fails to start on OpenSUSE Tumbleweed "Failed to initialize OpenGL loader!"

s0upsup, 2021-08-15:
Official package and building from source both result in the same error when trying to run.
"Failed to initialize OpenGL loader"

Now that I've built it, shouldn't be hard to provide additional info or try solutions if I can help out.  Project looks great btw, can't wait to try it!

Startup log:
```
./sdrpp -r ../root_dev/
[2021-08-14 23:12:51.049] [info] SDR++ v1.0.3
[2021-08-14 23:12:51.049] [info] Loading config
[2021-08-14 23:12:51.164] [error] Failed to initialize OpenGL loader!
```

Lesprilibre, 2021-08-15:
Maybe your version of OpenGL is out of date, probably due to your graphic card driver....
There is no fix, but still I'm having the same issue on my old laptop, I give a try with VirtualBox, not the easiest way for sure, but it works !

waterwin, 2021-08-15:
If the hardware can handle OpenGL newer then 2.1 and the supplier has made drivers for your OS the issue can be solved. Find the latest graphics driver or check what version of OpenGL your graphics card still supports.

s0upsup, 2021-08-15:
@Lesprilibre @waterwin Thanks for the help guys.  I haven't had issues with any graphical apps up until yesterday but I just tried a few and it seems possibly due to a system update or something, other programs are having trouble as well.  This does not seem to be sdrpp related but just bad timing with my system breaking :/
Sorry for the erroneous ticket!

closed 2021-08-15 by s0upsup

mnhauke, 2021-08-20:
> Official package and building from source both result in the same error when trying to run.
> "Failed to initialize OpenGL loader"
>
> Now that I've built it, shouldn't be hard to provide additional info [...]

The issue is related to a recent change in openSUSE Tumbleweed when they enabled support for EGL in glew ...
I just created a ticket: https://bugzilla.opensuse.org/show_bug.cgi?id=1189657

AlexandreRouma, 2021-08-20:
Thanks for finding this!
I don't known much about OpenSuse but if they need any help I'll be here.

mnhauke, 2021-08-20:
> Official package and building from source both result in the same error when trying to run.
> "Failed to initialize OpenGL loader"
>
> Now that I've built it, shouldn't be hard to provide additional info [...]

Until a proper fix could be supplied you could use the following glew package for openSUSE TW
https://build.opensuse.org/project/show/home:mnhauke:glew-without-egl

AlexandreRouma, 2021-08-20:
Is there anything I can do in code to work around this or is this an actual bug?

s0upsup, 2021-08-29:
Confirm, it's working after system repair/update. :+1: