AlexandreRouma/SDRPlusPlus #259: No audio or audio settings

ITCMD, 2021-08-15:
V 1.0.3
Windows 10

I believe I accidentally removed a module from module manager for controlling audio output. I see sdrplay_source and audio_sink in the list of active modules, but still no tab for audio settings and no output atm.

Where would audio settings be?

ITCMD, 2021-08-15:
[image]

AlexandreRouma, 2021-08-15:
In the sink menu, you need to switch from "None" to "Audio"

AlexandreRouma, 2021-08-21:
Closing for inactivity, please reopen if the above solution didn''t fix your issue

closed 2021-08-21 by AlexandreRouma

ITCMD, 2021-08-22:
Apologies for forgetting to reply. When I saw this I was away from the software, and by the time I fixed it I had forgotten. This did the trick!